riscv/riscv-isa-manual #894: base ISA naming conventions for E and I

neelgala, 2022-09-24:
I believe its legal for an implementation to have support for both : E and I base extensions. But section 28.2 in the spec says : 

```
RISC-V ISA strings begin with either RV32I, RV32E, RV64I, or RV128I indicating the supported address space size in bits for the base integer ISA.
```

Shouldn't RV32IE and RV64IE also be valid isa strings ?

aswaterman, 2022-09-24:
It is indeed legal for an implementation to support, for example, both RV32I and RV32E. But we consider these to be different base ISAs. E is not an extension of I, nor is I considered to be an extension of E, and so the ISA string RV32IE is invalid.

neelgala, 2022-09-24:
so what should be the legal ISA string for that device ? RV32I or RV32E ?

neelgala, 2022-09-24:
I ask because.. in the arch-test framework we choose the I or E tests based on which character is present in the ISA string (from riscv-config yaml). For devices that implement both, as per your suggestion, we might now end up running only of those extensions instead of both. So then we need a separate field in the YAML to indicate it both exist or not - which somehow feels unpleasant.

I know misa can be used to discover if "E" is implemented or not.. but misa itself is an optional CSR, and it falls under the privilege spec which is not a required dependency for "E" to exist.

aswaterman, 2022-09-24:
We describe such a hart as supporting both base ISAs. At any moment in time, it only uses one base ISA, and so it can correctly be described as either an RV32I hart or an RV32E hart, depending on its current mode. But we wouldn’t describe it as being both simultaneously.

I do understand the concern you’ve raised. Conceivably, we could invent a new extension name to indicate an RVI hart supports RVE mode via the misa register (Sme?). Similarly, we could invent a name to indicate that an RV64 hart supports being configured into RV32 mode (Smxt, for “XLEN three”?). But this is a half-baked idea and would need input from others, including e.g. our PrivArch chair @gfavor.

allenjbaum, 2022-09-25:
Strictly speaking, riscof doesn't need an ISA string for this, does it?
A hart may support both RV32I and RV64I, and could choose tests by
 -  looking in a specific directory
 -  or it could use the OR of two test_case macro conditions (i.e. RV64 or RV32)
 -  or even determine it because misa.XLEN is writable
Couldn't we do the same thing thing here?

pawks, 2022-09-25:
The ISA string in question here is the "ISA" field of the hart in the yaml file. 
The ISA string is the most reliable way to ensure that the test selection conditions are controlled inside the test itself. The current tests define a regex on the isa string. The xlen problem is already accounted for because every hart has a xlen field which is a list. So, if the hart support dynamic xlen switching it will have multiple entries in the list. 
The spec does make the following statement:
```
RV32E uses the same instruction-set encoding as RV32I, except that only registers x0–x15 are
provided. Any future standard extensions will not make use of the instruction bits freed up by the
reduced register-specifier fields and so these are designated for custom extensions.
```
So no standard extensions will be developed which are so to say "incompatible with I but compatible with E", meaning the problem can be solved by just having a boolean variable which indicates whether the `E` extension is supported and this variable can be set to `True` only if the base isa is `RV64I`. The caveat being that the above statement stands true for any "E" variants of the RV64 and RV128 base ISAs too.

neelgala, 2022-09-26:
@pawks is refering to the [supported_xlen](https://riscv-config.readthedocs.io/en/stable/yaml-specs.html#supported-xlen) field in the isa yaml which can be used to select rv32/64 tests for a device supporting both.

gfavor, 2022-10-11:
>
> Conceivably, we could invent a new extension name to indicate an RVI hart
> supports RVE mode via the misa register (Sme?). Similarly, we could invent
> a name to indicate that an RV64 hart supports being configured into RV32
> mode (Smxt, for “XLEN three”?).
>
The existing misa.I and misa.E bits have pretty solid description in the
Priv spec such that they are the enable bits for these two base
extensions.  Among other relevant sentences, the spec says:

*The “E” bit is read-only. Unless misa is all read-only zero, the “E” bit
always reads as the complement of the “I” bit. If an execution environment
supports both RV32E and RV32I, software can select RV32E by clearing the
“I” bit. *

Or is the issue more the need for an extension name to indicate whether the
I and E bits are fixed versus are configurable via misa.I writes?  And why
is this different than any other CSR bit in the architecture that can be
implemented as read-only or as writeable.   That just seems to go down a
path of naming every little thing in the architecture (including naming for
each WARL bit and field for the different possibilities of how it can be
implemented).

Practically speaking, couldn't or shouldn't the view for this issue be that
an implementation that supports multiple base extensions must separately
validate its compatibility with each base extension (and the differing
associated additional extensions supported on top of it).  The fact that
the implementation can support multiple base extensions is largely beside
the point.  Conceptually there is an RV'E' implementation and an RV'I'
implementation to be separately validated.  Plus validation that the misa.I
bit is writeable and that the misa.E bit is read-only !misa.I.

> Message ID: ***@***.***>
>

allenjbaum, 2022-10-11:
Although neither spec calls it out, I expect that a hart that supports both
I and E bases would always reset in I.
That probably should be explicit defined in the spec
   (the spec says that all extensions are enabled at reset, but that
doesn't work in this case).

If that is true, then we can run RV32I tests; in fact, it would be
perfectly legal to have an implementation that
executed instructions - including those using x16-31 identically (they're
reserved, so they can do anything they want).
In an implementation that does support both, those upper 16 registers are
actually there,
so passing the RV32I tests should be good enough to show that RV32E also
works for that hart.
The only additional test would be to show that misa.I can be reset and set,
with .E becoming the inverse.
Only when misa.I is readonly zero do we need to run a new set of tests.

aswaterman, 2022-10-11:
@allenjbaum "At reset [...] I shall be selected over E if both are available"

allenjbaum, 2022-10-12:
Ah, I looked in the E extension spec, and it wasn't there. I didn't look hard enough. mea-culpa.

closed 2025-12-09 by aswaterman