riscv/riscv-isa-manual #896: Confusing statements about the PMP behavior for M mode accesses.

pawks, 2022-09-26:
The spec says the following regarding the conditions where pmp checks are applicable:
```
Optionally, PMP checks may additionally
apply to M-mode accesses, in which case the PMP registers themselves are locked, so that even
M-mode software cannot change them until the hart is reset. In effect, PMP can grant permissions
to S and U modes, which by default have none, and can revoke permissions from M-mode, which
by default has full permissions.
```
So in theory, if none of the pmpcfg fields have their `L` bit set, no checks need to be performed on the M mode accesses.
However, in the further section describing the `Priority and Matching Logic`, the following statement is made:
```
The matching PMP entry must match
all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits. For example,
if a PMP entry is configured to match the four-byte range 0xC–0xF, then an 8-byte access to the
range 0x8–0xF will fail, assuming that PMP entry is the highest-priority entry that matches those
addresses.
```
So as per the second statement, all addresses should be checked for pmp matches first and unconditionally trap for a partial match.

Does the second statement take precedence over the first or is it legal to ignore partial matches if it is a M mode access without the L bit set?

scottj97, 2022-09-26:
The second statement must take precedent because you can't know if the L bit is set or not until you do the PMP lookup.

Keep in mind that an implementation may break up a large access into several smaller accesses in a way that makes it impossible for a single access to cross a PMP boundary. (E.g. a misaligned 8-byte LD to 0xffc could be split into two 4-byte accesses to 0xffc and 0x1000. It's not possible for either of those 4-byte accesses to cross a PMP boundary.) In such an implementation, if no PMP entry has the L bit set, then I don't see any need to do any PMP lookup for M-mode accesses.

pawks, 2022-09-26:
> The second statement must take precedent because you can't know if the L bit is set or not until you do the PMP lookup.

This statement assumes that the "pmp checks" referred to in the first statement are only the permission bits. But if we look at the "checks" to be the combination of both the match logic as well as the permissions then we can actually use the L bit to determine which pmps are applicable to the m mode access.  

> Keep in mind that an implementation may break up a large access into several smaller accesses in a way that makes it impossible for a single access to cross a PMP boundary. (E.g. a misaligned 8-byte [...]

I believe this is allowed only for misaligned accesses and not for aligned accesses. Is this assumption right?

scottj97, 2022-09-26:
> This statement assumes that the "pmp checks" referred to in the first statement are only the permission bits. But if we look at the "checks" to be the combination of both the match logic as well as [...]

I made no such assumption. The process goes:
1. Find which PMP entry(s) match the physical address range being accessed. (Assume a hidden, internal, lowest-priority PMP entry that matches all address space with L=0 and RWX=0.)
2. If that's more than one, take access fault
3. Calculate effective permissions from the one matching PMP entry: `L==0 && effective_privilege==M ? 3'b111 : RWX` (I.e., if the matching entry has L=1, the RWX bits apply to any access from any privilege mode. If L=0, M-mode has full permissions in this range.)

> I believe this is allowed only for misaligned accesses and not for aligned accesses. Is this assumption right?

I'm not sure. It's certainly allowed for misaligned.

gfavor, 2022-09-26:
Two corrections:

- An access may match multiple PMP entries - which is why they are
prioritized.  The highest priority match is considered to be the "match" -
the spec says "The lowest-numbered PMP entry that matches any byte of an
access determines whether that access succeeds or fails."  There is no
access fault simply because of multiple matches.

- If an access (whether naturally aligned or misaligned) "matches" two PMP
entries (i.e. part of the access "matches" one entry and another part of
the access "matches" another entry), then the spec says "The matching PMP
entry must match all bytes of an access, or the access fails."  That's when
an access fault comes into the picture.

BUT, to be more precise, the spec later says "On some implementations,
misaligned loads, stores, and instruction fetches may also be decomposed
into multiple accesses, some of which may succeed before an access-fault
exception occurs."  In other words, the "access" done by a load or store
may be different from the "access(es)" that is/are checked by PMPs.

scottj97, 2022-09-26:
> Two corrections:

My step 1 was implicitly assuming the priority was being considered, but I should have stated that.

And thanks for finding the text about decomposition, which answer's @pawks question about only allowed for misaligned.

pawks, 2022-09-26:
Thanks. That clears up things a lot.

The decomposition text now raises the question as to what if a particular misaligned instruction access matches 2 different pmp entries(both pmp entries allow the necessary permissions). Is it legal behavior for the implementation to trap or not trap in this case?

scottj97, 2022-09-26:
> Is it legal behavior for the implementation to trap or not trap in this case?

I believe either would be legal behavior. Trap, or not trap.

allenjbaum, 2022-09-26:
So, the same PMP configuration on two different implementations may have
very different behaviors.
E.g in this corner case where a misaligned access is matched by 2 different
entries, then a store might

   - succeed if the access is decomposed and both entries allow writes
   - write nothing.   , trap with cause "misaligned store" if the access is
   not decomposed , xtval pointing to effective address
   - write nothing,     trap with cause "access fault" if neither PMP entry
   allows writes, xtval pointing to upper half
   - write nothing,     trap with cause "access fault" if neither PMP entry
   allows writes, xtval pointing to lower half
   - write upper half, trap with cause "access fault" if lower   PMP entry
   disallows writes, xtval pointing to lower half
   - write lower half,  trap with cause "access fault" if lower   PMP entry
   disallows writes, xtval pointing to upper half

I'm not even sure you can depend on the same thing to happen each time you
execute something matching this condition.
With apologies to Bill:
I  intend that Sail will eventually be configured to select which of the 4
decomposed access trapping behaviors is implemented
 (and ignoring non-deterministic results; if that happens, the
implementation can apply for a waiver for that particular case).
I don't think that Sail support will happen any time soon; this is pretty
contrived

pawks, 2022-09-27:
No matter the granularity of the pmp entries in a platform, if the platform implements the C extension, it is inevitable that an instruction access matches 2 different pmp entries. Now since decomposition is allowed, this instruction access(a 4 byte access at a 2 byte boundary spanning 2 pmp regions) can still go through if both of the regions have execute access. But if it is legal for the mtvals to  point to the lower or upper half, then it is very difficult to recover from such a scenario. I believe the lower address should always be reported in the mtval if both accesses violate, otherwise the address of the violating access should be reported. 

The spec mentions the following line: `PMP violations are always trapped precisely at the processor.` The spec only allows the side-effects of other accesses(if they are legal) to be visible but the violations must be trapped precisely.

allenjbaum, 2022-09-27:
I don't think inevitable is the word I would use.
IT makes it possible, but highly unlikely.
"precisely", in this contains, says nothing about what is stored in xTVAL,
but when the exception is taken,
 e.g. before any architectural state of the core is modified (well, outside
the CSR rxception state...)
I think there may be explicit exceptions to that also, e.g. mstatus.fs
could be modified even if an LFD op trapped.

But the spec says:

 If mtval is written with a nonzero value when an ....
access-fault,...exception occurs on
an instruction ... store, then mtval will contain the faulting virtual
address.

I interpret that to mean the decomposed access that was caught by the PMP
entry,
which might be the lower address (base+offset), or the upper one
(base+offset+width-1)&(width-1)

aswaterman, 2025-12-09:
Closed as answered.