riscv/riscv-isa-manual #897: Is it allowed to prioritize debug entry and NMIs over interrupts reenabled following the execution of an xRET instruction or an explicit write to a CSR on which these interrupt trap conditions expressly depend?

Silabs-ArjanB, 2022-09-27:
The RISC-V Privileged spec version 20211203 states the following in section 3.1.9 Machine Interrupt Registers (mip and mie):

> These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time
> from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated
> immediately following the execution of an xRET instruction or an explicit write to a CSR on which
> these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).

It is our interpretation of the above text that it only states something about when the  conditions for an interrupt trap to occur must be **evaluated**, not when an interrupt trap must be **taken**. Specifically, in our implementation, even if the **condition** for an interrupt trap indicates that an interrupt trap is allowed to occur, there might be a higher priority event that prevents the interrupt from being **taken**. In our core such higher priority events are synchronous debug entry (e.g. due to trigger match), asynchronous debug entry (due to a halt request), and NMIs (custom).

To give a specific example of what could happen: Suppose a CSR write to the mie CSR re-enables an already pending interrupt. In that case (assuming the interrupt is also globally enabled) the condition for  the interrupt trap is immediately evaluated and becomes 'true' (allowing for the interrupt to be taken). However, if the next instruction (in the program right after the mie CSR write) is for example an instruction that has a trigger match, then in our core debug entry will occur (as opposed to interrupt entry).

Are our interpretation (described above) and stated example compliant with the RISC-V Privileged spec?

gfavor, 2022-09-27:
A few comments:

- The Debug 1.0 spec specifies the relative priority of debug exceptions.

- Very little about NMIs is currently architecturally specified (including
priority-related behaviors).

- That Priv quote is indicating that in general there can be some
non-trivial (but bounded) delay from when a hart sees an interrupt request
(and it is reflected in mip), to when the interrupt recognition logic sees
and responds to the interrupt request (aka takes the interrupt) at the next
moment that the logic evaluates all the specified conditions.  And the
quote is specifying the one exception where an interrupt request - that is
visible in mip - must be immediately recognized and taken after execution
of an xRET/etc.

- More pointedly to the question, "evaluation" and "taking" are considered
to go together.  But at that moment of evaluating and possibly deciding to
"take" an interrupt, the evaluation applies prioritization among the
eligible events to be taken.  Wrt to the quoted text, if a higher priority
event is taken, then upon return the lower priority interrupt request must
be taken before executing the next instruction after the xRET/etc.

> Message ID: ***@***.***>
>

Silabs-ArjanB, 2022-09-28:
@gfavor Thank you for your explanation. That's clear.

closed 2022-09-28 by Silabs-ArjanB