riscv/riscv-isa-manual #899: Unclear dependency relationship between H, S, and U

alexrp, 2022-10-02:
From a cursory glance through the privileged spec, I gathered that M is required while H, S, and U are optional. What's not clear to me is whether any of the latter 3 modes depend on one another. Intuitively, I would imagine that H depends on S which depends on U, but it isn't explicitly spelled out, so then again, maybe not?

Whatever the case may be, it seems like it might be worth spelling this out explicitly somewhere.

allenjbaum, 2022-10-03:
The legal combinations (which I thought we spelled out are:
M
MU
MUS
MUSH

No other combinations are legal as far as I know.

alexrp, 2022-10-03:
There is indeed a table in §1.2 "Privilege Levels" looking similar to that, but there is no mention of how H fits in there. That plus the overall wording in that section led me to believe that the table was only meant as a guideline.

FWIW, I think adding H into the table and renaming the "Supported Modes" column to "Required Modes" (or something to that effect) would make things clearer.

pdonahue-ventana, 2022-10-03:
> From a cursory glance through the privileged spec

You might want to look at table 1.2 and the text immediately below it, plus the beginning of the hypervisor extension as Scott mentioned.

alexrp, 2023-04-02:
Closing since the information seems to be there and I just missed it.

closed 2023-04-02 by alexrp