riscv/riscv-isa-manual #905: Unclear behaviour on vmid_bits detection in Bare translation mode

mingyuan-xia, 2022-10-26:
Quote the spec, section 8.2.10 on page 109, regarding `hgatp`. `hgatp` must be all zeros in Bare translation.

> When MODE=Bare, guest physical addresses are equal to supervisor physical addresses, and there is no further memory protection for a guest virtual machine beyond the physical memory protection scheme described in Section 3.7. In this case, the remaining fields in hgatp must be set to zeros.

and page 110, about the vmid_bits detection process, indicating that this detection can only be done in non-Bare modes since we need to csrw hgatp and read it back.

> The number of VMID bits is unspecified and may be zero. The number of implemented VMID bits, termed VMIDLEN, may be determined by writing one to every bit position in the VMID field, then reading back the value in hgatp to see which bit positions in the VMID field hold a one. The least-significant bits of VMID are implemented first: that is, if VMIDLEN > 0, VMID[VMIDLEN- 1:0] is writable. The maximal value of VMIDLEN, termed VMIDMAX, is 7 for Sv32x4 or 14 for Sv39x4, Sv48x4, and Sv57x4.

However, spike/qemu allow for Bare-mode vmid bits detection and [KVM relies on that](https://github.com/ultrarisc/linux/blob/master/arch/riscv/kvm/vmid.c#L23). In particular, a new hart boots and kvm detects vmid bits when the `old` hgatp is ZERO(Bare). Thus KVM works with spike/qemu yet violates the spec book.

We want to clarify the intended behaviours of vmid_bits detection in Bare translation mode. If `hgatp` is zero and we detect vmid bits, should the CPU raise an exception or disregard the write (remain 0) or return the actual vmid bits (spike/qemu way)?

scottj97, 2022-10-26:
There is also this statement in the spec:

> If hgatp.MODE is changed for a given VMID, an HFENCE.GVMA with rs1=x0 (and rs2 set to either x0 or the VMID) must be executed to order subsequent guest translations with the MODE change -- even if the old MODE or new MODE is Bare.

Which implies that VMID is significant even for MODE=Bare.

[Also @mingyuan-xia you should use a [permalink](https://github.com/torvalds/linux/blob/4dc12f37a8e98e1dca5521c14625c869537b50b6/arch/riscv/kvm/vmid.c#L23) when linking to a specific line of code.]

mingyuan-xia, 2022-10-27:
Yeah, the TLB flush is standard and the implication is valid. Yet, still feel that the three quotes are related but a bit complicated.
Would it be better to clarify that bare-mode hgatp must be zero when the translation is on after the TLB flush. Or, VMIDLEN detection can be done in any mode.

Permalink replaced, thx.