riscv/riscv-isa-manual #907: What's the definition of misa.B regarding to zb* extensions

Min-Yu, 2022-11-01:
For RISC-V Privileged Spec v. 20211203, the misa.B is labeled: _Tentatively reserved for Bit-Manipulation extension_.
For RISC-V Bit-Manipulation ISA-extensions Spec v1.0.0, defined four extensions (zba, zbb, zbc, and zbs).

For these specs, we can't find any description between misa.B and zb* extensions. 
We are confused about the setting condition of misa.B, for example:
1. Set misa.B when all zb* extensions are implemented.
2. Set misa.B when the basic bit-manipulation extension (zbb) is implemented.
3. misa.B shouldn't be set even if all zb* extensions are implemented.

aswaterman, 2022-11-01:
See https://github.com/riscv/riscv-isa-manual/commit/c1c77c4902d5e7c58613d725d09d66a2a743da1c

There is no relationship between misa.B and Zb*.  Currently, there is no way to control the presence of the Zb* extensions.

Min-Yu, 2022-11-13:
Thanks for your prompt reply

closed 2022-11-13 by Min-Yu