riscv/riscv-isa-manual #932: Shall WFI resume because of NMI?

Silabs-ArjanB, 2022-12-07:
It is clearly stated in the Privileged Specification that a WFI is "required to resume execution for locally enabled interrupts pending at any privilege level, regardless of the global interrupt enable at each privilege level". 

How about NMIs, are they _required_ to, or _allowed_ to, cause a resume of execution? I would assume that they are at least _allowed_ to cause a resume, but it is not clear to me whether they are _required_ to cause a resume.

aswaterman, 2022-12-07:
My reading of the spec is that, definitionally, NMIs are always locally enabled, because there are no mechanisms to locally disable an NMI.  (There's only a global NMI enable.)  So, the sentence you quoted requires that execution resume when an NMI becomes pending.

Silabs-ArjanB, 2022-12-08:
Okay, thank you

closed 2022-12-08 by Silabs-ArjanB