riscv/riscv-isa-manual #937: Clarification of mixed size atomic accesses

terrarier2111, 2022-12-13:
The A extension's documentation doesn't make clear whether mixed size atomic accesses are legal, illegal or explicitly states that it's unspecified.
Mixed size atomic accesses are atomic accesses/operations on a single memory location that vary in size from different harts e.g one hart accesses the memory location with atomic instructions that operate on full words and another hart accesses the same memory location using atomic instructions that operate on less than a full word.

aswaterman, 2022-12-13:
I think it follows from the axioms in the memory model that mixed-size
atomics are legal and do what you’d like. I don’t think any new normative
text is required here.

terrarier2111, 2022-12-14:
Thank you for the clarification, I can wasn't able to deduce this from the rules layed out by the spec but maybe that's just a personal problem. You can close this if you deem it unnecessary.