riscv/riscv-isa-manual #941: Clarification on misaligned AMO exception priority

whchiao1, 2022-12-21:
Privileged spec 1.12 relexed the exception priority of misaligned fault and page fault. Implementation could pick high priority misaligned fault or low priority misaligned fault. AMO without zam, misaligned AMO will issue access fault. However, according to exception priority table (Table 3.7), access fault priority is still lower than page fault, even it is coming from misaligned AMO. Should the priority be: access fault due to misaligned AMO (high priority implementation) > page fault or access fault during address translation > access fault of physical address > access fault due to misaligned AMO (low priority implementation)?

allenjbaum, 2022-12-22:
The version of the spec that I'm looking at (dated Sept 21, 2022) has just
that:

access fault due to misaligned AMO (high priority implementation) >
page fault *or access* fault during address translation >
access fault of physical address >
access fault due to misaligned AMO (low priority implementation)

So, an misaligned AMO could fault because the address translation itself
failed (e.g. wasn't allowed to access the PTE or PTE was invalid - which is
a page fault),
or the address translation succeeded - but it was misaligned
I'm not exactly sure which is chosen if the PTE entry doesn't have RW
permissions for the AMO
(is that considered part of the address translation, or part of the
explicit memory access with physical address?),
but it gives the same result either way.

aswaterman, 2025-12-09:
Closed as answered.