riscv/riscv-isa-manual #951: Hypervisor interrupt trap to HS mode

jrahmeh, 2023-01-06:
Section 9.2.3 of the privileged spec states:
```
An interrupt i will trap to HS-mode whenever all of the following are true: (a) either the current operating mode is HS-mode and the SIE bit in the sstatus register is set, or the current operating mode has less privilege than HS-mode; (b) bit i is set in both sip and sie, or in both hip and hie; and (c) bit i is not set in hideleg.
```
This seems to contradict the statement in section 3.1.8:
```
By default, all traps at any privilege level are handled in machine mode
```
Would it be possible to clarify the text in section 9.2.3 with something like:
```
An interrupt i, delegated to HS-mode in SIDELEG, will trap to HS-mode whenever ...
```

scottj97, 2023-01-06:
"By default" is a bit misleading, since it depends on `mideleg` and `medeleg` which have no specified reset value.

There is no `sideleg` CSR. I assume you meant `mideleg`?

The clarification you suggest is correct but not strictly necessary, since `sie`, `sip`, `hie`, and `hip` are all masked by `mideleg`. Any interrupt not delegated to HS-mode by `mideleg` will never appear in those CSRs, so clause (b) can never be true.

jrahmeh, 2023-01-06:
Thank you Scott. Closing issue 951.

closed 2023-01-06 by jrahmeh