riscv/riscv-isa-manual #961: WFI and zero time limit

JamesKenneyImperas, 2023-01-18:
Section 3.1.6.5 of the Privileged Architecture manual discusses the implementation-specific bounded time limit for WFI. It says:

> The time limit may always be 0, in which case WFI always causes an illegal instruction exception in less-privileged modes when TW=1.

Does this mean that an illegal instruction exception will occur when the time limit is 0 _even if a locally-enabled interrupt is pending but interrupts are disabled at the time the WFI is executed_? Had the implementation-specific time limit been _non-zero_ in this case, the WFI would not have caused a stall at all, because the wake up condition is already true. It feels like the specified behavior with time limit 0 makes that a special case which is not consistent with the non-zero time limit case. Can you please confirm the required behavior?

Thanks.

aswaterman, 2023-01-18:
Setting aside the confusion that might arise from coupling this behavior to the time limit: the dependent clause says pretty clearly that generating the exception unconditionally is a valid implementation.  (And of course since the time limit can be greater than 0, it's also valid to _not_ take the exception if there's an interrupt pending.)

allenjbaum, 2023-01-19:
I'm not sure how to interpret this.
If the option to unconditionally trap is implemented, the hart always
traps. I get that.
But in the case that
 - the unconditional trap option is not implemented,  and
 - the time limit==0 (is that possible if the trap option is not
implemented?)
 - exceptions are pending
 -  interrupts are not enabled
 - TW=1
should I assume that the implementation might trap, or might not trap?
If either is possible, is that deterministic behavior, or can it sometimes
trap and sometimes not?

This is at least one new item for the architectural option list (always
trap),
and perhaps more, so I need to know the range of possibilities here.

JamesKenneyImperas, 2023-01-24:
Hello @aswaterman,

> Setting aside the confusion that might arise from coupling this behavior to the time limit...

My point is that the specification _as written_ seems to couple `WFI` behavior to the time limit, and I was asking whether this was the intention or not. In very simple terms, WFI **when the time limit is zero** currently looks to be specified like this:

```
if trap_on_WFI() {
    take_illegal_instruction_or_virtual_instruction_trap()
} else if pending_globally_disabled_interrupt() {
    // don't stall
} else {
    stall()
}
```

Whereas the behavior **when the time limit is greater than zero** looks to be specified as:

```
if pending_globally_disabled_interrupt() {
    // don't stall
} else {
    if trap_on_WFI() {
        schedule_illegal_instruction_or_virtual_instruction_trap()
    }
    stall()
}
```

So the order of `trap_on_WFI` and `pending_globally_disabled_interrupt` tests seems reversed when the time limit is zero, which makes behavior inconsistent (in my view) when there is a pending globally disabled interrupt when `WFI` is executed.

It's quite likely I've entirely misunderstood, so apologies in advance if so, but the complexity of the behavior here is such that I think it would be very helpful to include pseudocode in the specification: I agree with @allenjbaum that it is not clear what the architectural possibilities are, and pseudocode would resolve this.

The primitives I've assumed to exist (`trap_on_WFI`, `take_illegal_instruction_or_virtual_instruction_trap`, `pending_globally_disabled_interrupt`, `schedule_illegal_instruction_or_virtual_instruction_trap`) are all quite complex as well: I've glossed over that here, but again pseudocode would help clarify things.

Thanks.

aswaterman, 2023-01-24:
Your understanding of the semantics matches mine. I think this can be resolved by improving the prose, though. There are plenty of more complicated concepts where we don’t resort to pseudocode; it seems disproportionate to do so here.

Let’s try to improve the description and see if you then agree. Do you want to take a stab first, or shall I?

JamesKenneyImperas, 2023-01-24:
Hi Andrew,

I think I'd prefer it if you had first go at this because it still isn't clear to me what the full range of architectural possibilities can be (and also I made a real dog's breakfast of my previous pull request, and this is much harder).

To confirm my understanding of your response here: you are saying that the two snippets I've written above describe the required behavior, even though the order of the two tests seems reversed - is that right?

Thanks.

aswaterman, 2023-01-24:
Yep.

I’ll take a stab at this tomorrow, hopefully.

aswaterman, 2023-01-27:
I want to request a consult from @jhauser-us.  The question is: is the phrasing "the time limit may be 0" sufficient to justify the notion that the exception may be raised unconditionally--even if a suitable interrupt is pending prior to the execution of the WFI?  The concern is that some readers might expect that, even if the time limit is 0, the implementation should check whether a suitable interrupt is pending before deciding to trap.  One fix would be to separate the unconditional-trap option from the time limit, with an edit like this:

```
diff --git a/src/machine.tex b/src/machine.tex
index 4733875..19889b6 100644
--- a/src/machine.tex
+++ b/src/machine.tex
@@ -866,10 +866,11 @@ When TW=0, the WFI instruction may execute in lower
 privilege modes when not prevented for some other reason.  When TW=1,
 then if WFI is executed in any less-privileged mode, and it does not complete
 within an implementation-specific, bounded time limit, the WFI instruction
-causes an illegal instruction exception.  The time limit may always be 0, in which
-case WFI always causes an illegal instruction exception in less-privileged modes
-when TW=1.  TW is read-only 0 when there are no modes less privileged than
-M.
+causes an illegal instruction exception.
+Implementations are also permitted to unconditionally raise an illegal
+instruction exception when WFI is executed in a less-privileged mode when
+TW=1.
+TW is read-only 0 when there are no modes less privileged than M.
 
 \begin{commentary}
 Trapping the WFI
```

Note, if we do change the text, we also need to change the hypervisor spec in multiple places.

jhauser-us, 2023-01-30:
> The time limit may always be 0, in which case WFI always causes an illegal instruction exception in less-privileged modes when TW=1.

As I read the sentence above, it clearly implies, *if* TW = 1 and a given implementation has a time limit of 0 for this case, *then* an attempt to execute a WFI in a less-privileged mode always raises an illegal instruction exception.  That is, the sentence asserts by fiat:

TW = 1 AND time limit = 0  ->  WFI always traps

James Kenney wrote:
> It feels like the specified behavior with time limit 0 makes that a special case which is not consistent with the non-zero time limit case.

I agree.  But that's what the document says.

Allen Baum asked:
> But in the case that
> - the unconditional trap option is not implemented,  and
> - the time limit==0 (is that possible if the trap option is not implemented?)

I believe this combination is not allowed, because time limit = 0 implies unconditional trapping.  However, you can get the same effect by assuming the time limit is vanishingly small but still greater than 0.  E.g. on my processor, I could say the time limit is 1 femtosecond, which these days is less than 1/100,000th of the amortized time to execute one instruction.  So in practice an implementation can choose either:  (a) unconditionally trap; (b) adopt the non-zero time limit behavior Kenney suggested (with a suitably small time limit); or (c) have a visibily larger time limit.  Any of the three is allowed.

Should unconditional trapping be one of the allowed choices?  I think the current text is clear enough that the question is probably moot for the ratified spec.  Unconditional trapping *is* allowed.  Eliminating the option would be a normative change, not just a clarification.

allenjbaum, 2023-01-30:
In reality, anything less than one cycle is probably equivalent to zero,
for any definition of cycle counted by the cycle counter.
(effectively, it's a Planck second for the purposes of processor time) So,
1 femtosecond==0,
IF you believe that I would then say you can't get the same effect with a
vanishingly small time limit.

My conclusion is that if unconditional trap is implemented, then the time
limit>0 (and you conditionally trap)
 That leaves 2 legal scenarios:
*unconditional trap    **timeout-limit *
   yes                              ---          unconditionally trap
    no                                >0           trap when timeout limit
is exceeded (so will trap if interrupt is already pending
  (no                              ==0            not allowed)

 so for any legal configuration

(which riscv-config should check,  and therefore the YAML configuration
file it uses must specify
both "unconditionally-trap"  *and* some timeout llmit that a test can
guarantee in terms of instret)

 - unconditionally trap will trap unconditionally regardless of timeout
limit
 - otherwise, it will trap if no interrupt is pending within the timeout
limit

JamesKenneyImperas, 2023-01-30:
It seems that the conclusion of this thread is that behavior with time limit zero is intentionally inconsistent with the case when the time limit is non zero, so perhaps a tweak to the sentence in 3.1.6.5 to make this explicit would be helpful:

> The time limit may ~~always~~ be 0, in which case WFI always causes an illegal instruction exception in less-privileged modes when TW=1, **even if there are pending globally-disabled interrupts when the instruction is executed**.

(I suggest removing the first _always_ from this sentence, as to me this implies that in some implementations the time limit may sometimes be zero and sometimes non-zero.)

The use of vanishingly small delays as a device to permit implementations that effectively have a zero time limit to not stall when there is a pending globally-disabled interrupt is very non-obvious to me. If this is really an implementation option then I think there needs to be an explicit note about it.

jhauser-us, 2023-01-30:
Even better, I think we could also remove from the problem sentence any mention of the time limit being zero, and just say:
> An implementation may have WFI always raise an illegal instruction exception in less-privileged modes when TW=1, even if there are pending globally-disabled interrupts when the instruction is executed.

aswaterman, 2023-01-30:
This change is basically what I was driving at, so naturally I agree. I’ll take care of it and close the issue. Thanks, everyone!

aswaterman, 2023-01-30:
@jhauser-us In that case, don't we also need to revisit the definition of VTW?

There is also the mention of VTW towards the end of the chapter in the list of virtual instruction exception causes, but I think that text is OK as-is.

I'll make a PR and wait for your approval.

allenjbaum, 2023-01-30:
Looking at the Imperas configuration options (which correspond to
architectural o[ptions not controlled by CSR bits, and there may be
implementations that set them differently), they have:
wfi_is_nop.  - this is mentioned in the spec already
wfi time limit - this is mentioned in the spec already implementation
dependent both as amount and as type (how is the interval measured?)
wfi_0limit_behavior - should this be renamed as: trap_always (well, in a
lower priv mode with TW=1) since it no longer cares what the timeout limit
is?

I don't see an option for "wfi-umode_support", which the spec mentions:
    "WFI is available in all privileged modes, and optionally available to
U-mode."
Or does this introduce new behavior that needs another option to be
defined? I suspect - and am hoping, it is the former

I'd love to see an equation which describes the behavior under all
combinations of priv modes, locall/global enables, interrupt pending, and
the 4 architectural options above

allenjbaum, 2023-01-30:
Also: if WFI is unavailable in U mode, and the "always treat as noop" is
defined - should it trap, or should it noop? Or is that combination illegal?

closed 2023-01-31 by aswaterman

JamesKenneyImperas, 2023-01-31:
Hello @allenjbaum - I agree with you that a summary table of the possible behaviours would be really useful.

The Imperas model originally had just the `wfi_is_nop` parameter as you describe and assumed a time limit of zero. We've now added a `TW_time_limit` parameter to allow a non-zero cycle delay to be specified (this is also used by the `wrs.nto` instruction in the `Zawrs` extension, if that is configured). The time limit is only used for `WFI` when `wfi_is_nop` is `False` (because when  `wfi_is_nop` is `True` there is obviously no stall to time out).

Given the above specification update, we will add another Boolean parameter (`wfi_resume_not_trap`, perhaps?) to specify whether, when `TW_time_limit` is zero or `wfi_is_nop` is `True`, an active resume event when `WFI` is executed prevents a trap occurring if it otherwise would.

Assuming these parameter names, I think distinct configuration options are:

```
   wfi_is_nop    TW_time_limit    wfi_resume_not_trap   comment
   ----------    -------------    -------------------   ----------------------------------------------------------
1. True          N/A              False                 Trap if subject to TW=1 or U-mode with S-mode, else NOP.
2. True          N/A              True                  As case (1), but NOP if wakeup event already pending.
3. False         0                False                 Trap if subject to TW=1 or U-mode with S-mode, else stall.
4. False         0                True                  As case (3), but NOP if wakeup event already pending.
5. False         >0               N/A                   Stall until wakeup event or until TW=1 limit (if active);
                                                        resume if wakeup event, trap if limit reached.
```

(the comments are a bit brief, but hopefully enough to describe what I think the behaviour should be.)

You also said:

> I don't see an option for "wfi-umode_support", which the spec mentions:
>     "WFI is available in all privileged modes, and optionally available to U-mode."
> Or does this introduce new behavior that needs another option to be defined? I suspect - and am hoping, it is the former

Section 3.1.6.5 says:

> When S-mode is implemented, then executing `WFI` in U-mode causes an illegal instruction exception, unless it completes within an implementation-specific, bounded time limit. A future revision of this specification might add a feature that allows S-mode to selectively permit WFI in U-mode. Such a feature would only be active when TW=0.

So the behaviour when S-mode is implemented is as if an implicit S-mode `TW=1`: there is no new option to be defined there. If S-mode is _not_ implemented, then behaviour in U-mode is dependent on `mstatus.TW`, as we've discussed in this thread, but again there is no new configuration option needed for this, unless I've misunderstood something.

> Also: if WFI is unavailable in U mode, and the "always treat as noop" is defined - should it trap, or should it noop? Or is that combination illegal?

I believe that trap checking takes priority. So if following the rules described in sections 3.1.6.5 and 3.3.3 indicates that the `WFI` will trap, then it will trap, irrespective of whether the implementation is a `NOP`.

If I've got anything wrong here then hopefully @aswaterman or @jhauser-us can correct me.

allenjbaum, 2023-01-31:
comments inline

> Assuming these parameter names, I think distinct configuration options are:
>
>    wfi_is_nop    TW_time_limit    wfi_resume_not_trap   comment
>    ----------    -------------    -------------------   ----------------------------------------------------------
> 1. True          N/A              False                 Trap if subject to TW=1 or U-mode with S-mode, else NOP.
> 2. True          N/A              True                  As case (1), but NOP if wakeup event already pending.
> 3. False         0                False                 Trap if subject to TW=1 or U-mode with S-mode, else stall.
> 4. False         0                True                  As case (3), but NOP if wakeup event already pending.
> 5. False         >0               N/A                   Stall until wakeup event or until TW=1 limit (if active);                                                        resume if wakeup event, trap if limit reached
>
> Ugh; that's pretty ugly, so 5 different behaviors based on hidden
variables.; I'm not sure how code will be able to use this.

> You also said:
>
> I don't see an option for "wfi-umode_support", which the spec mentions:
> "WFI is available in all privileged modes, and optionally available to
> U-mode."
> Or does this introduce new behavior that needs another option to be
> defined? I suspect - and am hoping, it is the former
>
> Section 3.1.6.5 says:
>
> When S-mode is implemented, then executing WFI in U-mode causes an
> illegal instruction exception, unless it completes within an
> implementation-specific, bounded time limit. A future revision of this
> specification might add a feature that allows S-mode to selectively permit
> WFI in U-mode. Such a feature would only be active when TW=0.
>
> I was aware of that, but wasn't sure the "optionally available" was a
stand-alone option, or was referring to this. This text really needs to
point to sec 3.1.6.5

> Also: if WFI is unavailable in U mode, and the "always treat as noop" is
> defined - should it trap, or should it noop? Or is that combination illegal?
>
> I believe that trap checking takes priority. So [...]
I don't disbelieve you, but I don't see anything in the spec that is clear
about which would take priority. I am actually of the opinion that "always
noop" would take priority.
This requires spec clarification (unless you can point to something that is
unambiguously one way or the other). I can point to this, which simply
says, without an conditionality:

The purpose of the WFI instruction is to provide a hint to the
implementation, and so a legal implementation is to simply implement WFI as
a NOP.

That's a pretty general, easily understood statement, so in the absence of
anything else, I would conclude that it takes priority.
The earlier text says: (emphasis mine)

This instruction *may* raise an illegal instruction exception when TW=1 in
mstatus, as described in Section 3.1.6.5.

Which you then have to hunt down and figure out what it really means.  But
I see no explicit reason that the noop statement should not ovrride
regardless of earlier text .
If the earlier text would take priority, then this statement must be
modifed with exactly the conditions under which a legal implementation must
not NOP.

If I've got anything wrong here then hopefully @aswaterman

PR #972 (2023-01-31): Clarify WFI trapping behavior
Resolves #961.

See discusson on that issue for motivation.
commit c0b298a2 (2023-01-31): Clarify WFI trapping behavior (#972)

Resolves #961.

See discusson on that issue for motivation.
commit 07feb44e (2023-02-28): Clarify WFI trapping behavior (#972)

Resolves #961.

See discusson on that issue for motivation.