riscv/riscv-isa-manual #979: [question] about Zfa fcvmod.w.d instruction

Cosecant11, 2023-02-02:
Section 25.4 of the Unprivileged ISA manual has this note:
>Bits 31:0 are taken from the rounded, unbounded two’s complement result, then sign-extended to XLEN bits and written to integer register rd.
What is the meaning of unbounded?
And if the input number is +2^31,  XLEN=64, the result should be 64'hffff_ffff_8000_0000 or 64'h0000_0000_8000_0000?

aswaterman, 2023-02-02:
Unbounded means no limit on range.

Sign extension means bit 31 is copied into bits 63-32, so 64'hffff_ffff_8000_0000.

aswaterman, 2023-02-06:
In the standard RV64 ABIs, all 32-bit numbers are sign-extended to 64 bits, regardless of whether the 32-bit type is signed. Read the RV64 base ISA chapter for additional context.

aswaterman, 2023-02-06:
I should have also added: 2^31 is not representable as a 32-bit two’s complement integer. The bit pattern that corresponds to 2^31 is the one that represents -2^31.

aswaterman, 2023-02-06:
Think of it as taking the low-order 32 bits then sign-extending the result to 64 bits. Whatever that value happens to be is what the instruction produces.

Cosecant11, 2023-02-06:
Thanks for your answer

Cosecant11, 2023-03-27:
If the number(a double-precision value) is outside the range of signed int64, would the result be like the number mod 2^32? and is there any C model available for reference on the result of the instruction?

Thanks.

asb, 2023-03-27:
Patches for qemu [were posted this morning](https://lore.kernel.org/qemu-devel/20230327080026.3266574-1-christoph.muellner@vrull.eu/). I haven't seen anything for Sail or Spike yet.

davidharrishmc, 2024-01-17:
I have a question about flags set by FCVTMOD.W.D in the Zfa extension.  Specifically, consider
RV32IFD_Zfa running fcvtmod.w.d on an input of 0xC1EE9B7E5FC9EBA4 = -4.10806e9.
The integer output is 0x0B240D02. Should this set the inexact or invalid flag?

According to the Zfa spec at Section 26.4,
"Floating-point exception flags are raised the same as they would be for FCVT.W.D with the same input operand.”

According to the unprivileged spec at 12.5 for fcvt.w.d
"The range of valid inputs for FCVT.int.D and the behavior for invalid inputs are the same as for FCVT.int.S.”

And according to the unprivileged spec at 11.7 for fcvt.w.s
"If the rounded result is not representable in the destination format, it is clipped to the nearest value and the invalid flag is set. “

The rounded result is outside the range [-2^31, 2^31-1] so fcvt.w.d should set the invalid flag.
However, on the riscv-arch-test PR330 fcvtmod.w.d_b22-01 test at address 0x80000570-4, the fcvtmod.w.d sets fcsr = 0x00000001, setting
the inexact bit rather than the invalid bit.

This appears to be a bug in Sail.  Alternatively, the Zfa fcvtmod.w.d spec might be wrong/misleading because it is strange to set the invalid flag when the result after
moduo 2^32 is a valid integer.

Before I submit a Sail bug report, I’d like to confirm my understanding is correct.

aswaterman, 2024-01-18:
Take the ISA spec at face value, even though the specified behavior seems unusual.  (FWIW, Spike does correctly set NV rather than NX for out-of-range inputs.)

closed 2025-12-09 by aswaterman