riscv/riscv-isa-manual #981: V spec should be updated from Draft 0.7 to Frozen 1.0

hdelassus, 2023-02-09:
From the [v-spec 1.0](https://github.com/riscv/riscv-v-spec/blob/v1.0/v-spec.adoc#introduction):
> <html>
<body>
<div dir="auto" style="box-sizing: border-box; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px;">This document is version 1.0 of the RISC-V vector extension for public review.</p></div><div dir="auto" style="box-sizing: border-box; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Note | This version 1.0 is considered frozen for public review as part of the RISC-V International ratification process. Version 1.0 is considered stable enough to begin developing toolchains, functional simulators, and implementations, including in upstream software projects, and is not expected to have incompatible changes except if serious issues are discovered during ratification. Once ratified, the spec will be given version 2.0.
-- | --

</div><div dir="auto" style="box-sizing: border-box; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><br class="Apple-interchange-newline">
</body>
</html>

aswaterman, 2023-02-09:
Thanks for pointing this out.  But it's better than frozen; it's actually ratified!

closed 2023-02-09 by aswaterman

hdelassus, 2023-02-09:
Well, I'm not 100% sure, but it seems you were right: the spec seems to be ratified! From these posts:
* [RISC-V Announcement: RISC-V International Ratifies 15 New Specifications, Opening Up New Possibilities for RISC-V Designs](https://riscv.org/announcements/2021/12/riscv-ratifies-15-new-specifications/)
* [RISC-V Wiki: Recently Ratified Extensions](https://wiki.riscv.org/display/HOME/Recently+Ratified+Extensions)

The riscv-v-spec repo would deserve a refresher in its description and README :)

+ [`src/v.tex`](https://github.com/riscv/riscv-isa-manual/blob/master/src/v.tex) in this repo, too! :)

aswaterman, 2023-02-09:
Ah, nice.  Thanks.  I'll take care of all this.

aswaterman, 2023-02-09:
Conclusion: what was ratified is the frozen 1.0 spec, plus the responses to the public review feedback.  The latter has only been incorporated in to the spec in draft form.  Obviously we need to get a new version of the spec out the door and bless that as the ratified version.  I'll wait to update this document until we get to that point.  But I will fix the other reference to 0.7 you pointed out.

commit ad5afcbc (2023-02-09): V extension is frozen

Resolves #981