riscv/riscv-isa-manual #983: Question about CSRRSI and CSRRCI instructions

ivangaminoUAH, 2023-02-22:
Hi,

I have a couple of questions about the definition of the following instructions (taken form Volume I: RISC-V Unprivileged ISA V20191213):

[image]

Is there a reason for setting or clearing only the least 5 significant bits of a CSR?
Would it not be more practical to encode on those 5 bits the position (0 up to 31) of a bit to be set or cleared with CSRRSI and CSRRCI instructions?

Thanks in advance.

kasanovic, 2023-02-23:
The current definition allows multibit fields to be modified atomically , and keeps the same immediate datapath for RW and RS/RC operations.  They also support encoding of read-only CSR instructions. The definition of the current instructions is ratified, so new behavior would require defining new instructions with a rationale.

ivangaminoUAH, 2023-03-03:
First of all, thank you for your rapid response.

As the specification is curreltly ratified, there is no use in trying to modify it. The original purpose of this issue is trying to understand why there are instructions to  **exclusively** modify the lower 5 bits of a CSR instead of any other higher bit.

Thank you again for your attention.

closed 2025-08-04 by aswaterman