riscv/riscv-isa-manual #984: About the sc.w offset in Figure A.4

DehengYang, 2023-02-22:
Dear Authors,

Sorry for my disturbance.

I noticed that in the code snippets presented in Figure A.4, see https://github.com/riscv/riscv-isa-manual/blob/eae4f007ba6984058e0f08f69eba524b73ec083f/src/memory.tex#L306

```
  (a) lr.w a0, 0(s0)
  (b) sw   t1, 4(s0)
  (c) sc.w t2, 8(s0)
```

Seemingly `8` indicates an offset, but `sc.w` instruction has no offset. Furthermore, the paired `lr.w` and `sc.w`  are supposed to operate on the same address. **Therefore, is `8` in this instruction potentially a typo? It would be sincerely appreciated if any advice could be provided. Thank you!**

Yours sincerely,
Deheng

aswaterman, 2023-02-22:
It appears to be intentional. You’re right that the instruction isn’t legal in RISC-V, but this is essentially just pseudocode. (It’s also omitting the destination register for the `sc.w` result.) Pretend there’s an extra `addi` to generate the effective address and it makes sense again.

cc @daniellustig

daniellustig, 2023-02-22:
+1 to what Andrew said, sorry for the imprecise assembly in the examples

DehengYang, 2023-02-23:
Dear @aswaterman , @daniellustig 

Thank you so much for the prompt reply. That greatly helps!

I am a big fan of RISC-V ISA and its associated manual. While the omission of the destination register (`rd`) of the `sc.w` is perfectly understandable, the omission of the description about where `8` offset comes from may _cause potential confusion for the manual readers or users_. Therefore, **I create a PR that eliminates the potential confusion of subsequent readers (see PR https://github.com/riscv/riscv-isa-manual/pull/985). The PR makes a minimal change while not negatively impacting the original layout of the manual. Please kindly advise. Thank you!**

Thank you again for your time and help. Many thanks and wish you a great day!

Yours sincerely,
Deheng

aswaterman, 2023-02-23:
Let's move the discussion to #985

DehengYang, 2023-02-24:
Thank you again for the kind guidance and huge support!

closed 2023-02-24 by DehengYang