riscv/riscv-isa-manual #988: Question regarding reporting on instruction-address-misaligned exceptions

nmeum, 2023-03-09:
Volume I (User-Level ISA) states the following regarding the “instruction-address-misaligned exception”:

https://github.com/riscv/riscv-isa-manual/blob/2713791c03c8943656dc6f71a39c5dd0916f5260/src/rv32.tex#L153-L158

I am specifically interested in the interpretation of the following sentence from this paragraph:

> This exception is reported on the branch or jump instruction, not on the target instruction.

What does “reported on” mean in this context?

This requirement was added with commit db0b5ca0c2269ea494a2e9cf442b14d3039b4ce4, unfortunately without further explanation (CC: @kasanovic).

---

Taking Volume II (Privileged Architecture) into account: Does this mean that `mtval` should be set to the address of the branch or jump instruction? Because if so, this is somewhat contradictory to the following statement from the privileged architecture:

https://github.com/riscv/riscv-isa-manual/blob/b6480da0db0c428a8ab29b65cae693041cd3e874/src/machine.tex#L2213-L2216

P.S.: Volume I uses the spelling “instruction-address” and „instruction-address-misaligned exception” while Volume II uses “instruction address” as well as “instruction address misaligned exception” which is a minor inconsistency.

aswaterman, 2023-03-09:
"reported on" means that the value recorded in epc is the address of the branch or jump, rather than the address of the target.

aswaterman, 2025-12-09:
Closed as answered.

closed 2025-12-09 by aswaterman