riscv/riscv-isa-manual #989: Up-to-date link-able RISC-V spec?

sethp, 2023-03-11:
First off, thank you so much for the effort you've put into the RISC-V specs! They're absolutely a joy to read, and I really appreciate including the various commentary about why various decisions were made. As a neophyte to ISA design, I find those discussions of the possible tradeoffs incredibly helpful.

I'm using a writing project as an excuse to learn more about the RISC-V ISA, and I discovered today that I've been working based on a somewhat outdated version of the specs from 2017: when I google "riscv spec" the second result is a link to https://riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf

[image]

Also, for completeness: I've had the same experience with "riscv privileged spec" offering me a path to https://riscv.org/wp-content/uploads/2017/05/riscv-privileged-v1.10.pdf .

I've been using the 2017 one there primarily as I've discovered that the PDF viewers built into browsers (well, at least the Safari, Chrome, and Firefox that I tested) support fragment linking: e.g. https://riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf#section.1.1 or https://riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf#page=42 .  That allows me to leave a specific link in the document, so I can jump straight to the relevant section for ease of reference.

So, my question: is there a stable URL that offers a more recent version of the spec with `content-type: application/pdf` so my browser knows to display its contents? All I can find at the moment for up-to-date specs are `content-type: application/octet-stream`, which my browser downloads and opens, but loses the link fragment in the process. And, a bit spicier of a question: what do you think of updating the `riscv.org` robots.txt to disallow crawling `/wp-content/uploads/2017/05`?

gfavor, 2023-03-11:
> [image: a screenshot of google results showing the top two results, with
> the second as indicated]

Not a direct link, but this home page for the ISA manuals (
https://github.com/riscv/riscv-isa-manual) has a link, if you look down the
page, that always points to the latest release of both manuals.

sethp, 2023-03-11:
Thanks for the suggestion! I took a look through there, and all I could find were ultimately various links to the github release assets, which (at least all the ones I tried) all are being served as `content-type: application/octet-stream` (i.e. as downloads, not something that "opens in browser"). Did I miss something?

sethp, 2023-03-11:
I noticed that the links from `riscv.org` also look like they're going straight to the github release assets, too. The [docs](https://docs.github.com/en/rest/releases/assets?apiVersion=2022-11-28#upload-a-release-asset) seem to suggest it ought to be possible, at some level, to control what content type an asset is served with. Unfortunately, the docs are at least wrong about the expected request format, and once I stopped getting 400 errors I still wasn't able to get it to respect the asset's `application/pdf` content type. I opened a support ticket to get some eyes from GitHub on what their API should be doing, and I'll let you know if that goes anywhere useful.

In the meantime, I'll probably throw a recent version up in a public backblaze bucket or something (it's not like there's a shortage of internet-connected object stores in this day and age). Then, in 2029 someone can experience the joy of asking "why have I been looking at something from early 2023 this whole time?" :upside_down_face:

gfavor, 2023-03-11:
No you didn't.  The linked page contains links to the latest release of
each of the two ISA manuals as PDF documents.

sethp, 2023-05-01:
> The [docs](https://docs.github.com/en/rest/releases/assets?apiVersion=2022-11-28#upload-a-release-asset) seem to suggest it ought to be possible, at some level, to control what content type an asset is served with.

Just wanted to follow up on this: I heard back from the support folks, and they said:

> Due to the way Release assets are stored, users won't get back the original content type when retrieving the asset.

So it seems the content-type there is just a metadata field, not something that can be used to mark e.g. a PDF as a PDF instead of a generic byte stream.

I've gone ahead and uploaded the current latest "ratified" specs to a bucket that does serve them with the more specific content type, they're available here:

* [Volume 1, Unprivileged Specification version 20191213](https://f004.backblazeb2.com/file/shared-sethp/riscv-spec-20191213-2.pdf)
* [Volume 2, Privileged Specification version 20211203](https://f004.backblazeb2.com/file/shared-sethp/riscv-privileged-20211203.pdf)

I'd still encourage y'all to update the robots.txt to disallow crawling the `/wp-content/` path and consider an alternative hosting arrangement, at least for the ratified versions. I keep finding myself accidentally landing on the spec from six years ago via google searches, and it seems a shame that I'm missing out on the benefits of y'all's excellent efforts during that time.

Timmmm, 2025-02-26:
It's HTML but you can use these until there is a proper place where they are hosted:

https://riscv-specs.timhutt.co.uk/

I could put PDFs there if people want, but I find the HTML much easier to use.

Timmmm, 2025-02-26:
>  They shouldn't change.

Hmm I would be mildly surprised if you are able to maintain a link like https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view?usp=drive_link in the long term. Guess we'll see!