riscv/riscv-isa-manual #1006: Conflict in setting MPP for MRET

liweiwei90, 2023-04-06:
In section 3.1.6.1, the privilege spec says:
```
An MRET or SRET instruction is used to return from a trap in M-mode or S-mode respectively.
When executing an xRET instruction, supposing xPP holds the value y, xIE is set to xPIE; the
privilege mode is changed to y; xPIE is set to 1; and xPP is set to the least-privileged supported
mode (U if U-mode is implemented, else M). If y̸ =M, xRET also sets MPRV=0.
```
Which means MPP will be set to PRV_M if U-mode is implemented.

However, in section 9.6.4, it directly says:
```
MRET then in mstatus/mstatush sets MPV=0, MPP=0, MIE=MPIE, and MPIE=1.
```

pdonahue-ventana, 2023-04-06:
The spec says:
> xPP is set to the least-privileged supported mode (U if U-mode is implemented

You said:
> Which means MPP will be set to PRV_M if U-mode is implemented.

That's not right.  If U-mode is implemented then it's set to U-mode as it says.

liweiwei90, 2023-04-06:
Sorry for the typo, it should be "if U-mode is not implemented"

liweiwei90, 2023-04-06:
Maybe not a conflict.  I just find section 9 is for Hypervisor extension. If we assume hypervisor is supported for the description in this section, then S-mode, and U-mode should be supported too.  So directly set MPP to zero seems right in this case.

closed 2023-04-06 by liweiwei90