riscv/riscv-isa-manual #1007: If XLEN = 64 and address width is less than 64, shoud pc be signed-extend or zero-extend for JALR when writing RD？

jinyinghan324, 2023-04-07:
For example, if in our implementation, address width is 40, when PC is 0xFF_0000_0000，should RD of JALR  be 0xFFFF_FFFF_0000_0004 or  0x0000_00FF_0000_0004？

allenjbaum, 2023-04-07:
From Mmode, it's a physical address and is zero extended to the width of
your physical address
else from S or Umode, then
        if in bare mode: zero extended to the width of your physical address
else if VM is enabled, then signed extended to the width of whichever SV
mode you're in (SV32, SV39, SV48, SV57)
(actually, RV64 can't choose SV32...)

scottj97, 2023-04-07:
IIUC, in RV64, the PC is always conceptually a 64-bit value. If you try to put an out-of-bounds address into this PC, you get an access fault trap.

What's out-of-bounds? In M-mode or Bare translation mode, any PC exceeding the physical memory bus size. Effectively, zero-extended.

In Sv39 and other translation modes, any PC that's not a sign-extended 39-bit (or 48, or 57) value.

pdonahue-ventana, 2023-04-07:
I agree with everything @scottj97 said with the minor addendum that "If you try to put an out-of-bounds address into this PC, you get an access fault trap *in M-mode or Bare mode or a page fault trap otherwise.*"

allenjbaum, 2023-04-08:
The specific question was the other way around: what is the PC  value that
is put into a link register.

The spec doesn't explicitly say that PC+4 doesn't wrap, but I would guess
that a consensus is that it is implied.

I can certainly imagine that PC incrementer would be implemented to be just
the length of the  largest implemented address
( which varies depending on VA and PA),since that value is the only real
corner case.
so making it explicit that an address incrementer must generate a carry
past the MSB, rather than wrap would be good.
The Sail code would make that explicit.

gfavor, 2023-04-08:
The question of what value goes into a link register is orthogonal to memory access control matters such as what happens when an actual memory access uses an "out-of-bounds" address.

In the former case, irrespective of what are "good" or "bad" addresses, the full proper next sequential PC value is put into the link register (with a 64-bit wrapround if the call instruction is at the very end of 64-bit VA space).  No memory access is being performed by the call instruction with that address.

In the latter case what results, architecturally, in an Access Fault on an "out-of-bounds" address is the failing of PMA checks.  In essence one can imagine a default PMA region covering all 64 bits of PA space, with other explicit PMA regions (software configurable or not) covering parts of PA space and overriding the default region's attributes.  The default PMA region would be a "Vacant" region - which always results in an Access Fault.  As a result, all accesses that don't fall in one of the explicit PMA regions would take an Access Fault.

And in practice it would just so happen that the hardware for explicit PMA regions would only support regions within the low "implemented" part of PA space.  And any "out-of-bounds" addresses by definition would be covered by the default "Vacant" region.

pdonahue-ventana, 2023-04-08:
> The spec doesn't explicitly say that PC+4 doesn't wrap, but I would guess
that a consensus is that it is implied.

It does explicitly say that:
> memory address computations done by the hardware ignore overflow and instead wrap around modulo 2^XLEN.

The specific question above was not about wrapping.  The question is if PC=0xFF_0000_0000 then should the link register be 0xFFFF_FFFF_0000_0004 or 0x0000_00FF_0000_0004.  Obviously, 0xFF_0000_0000+4=0x0000_00FF_0000_0004 so I think that the question itself showed confusion about upper bits of the PC before the add.

aswaterman, 2023-04-08:
It comes down to what Scott wrote: the PC is always XLEN bits wide. Any uarch optimizations that compress the PC mustn’t be architecturally visible, including how the link register is manipulated.

allenjbaum, 2023-04-08:
Ah - I only looked at the definition of JAL which doesn't have the
statement about memory address computations,
It does say PC+4, but doesn't explicitly say what happens on overflow, or
what the overflow point is for wrap (though, as pointed out, it does have
that memory computation comment...somewhere).
It's likely JAL uses a different adder from the one that computes the
branch/olad/store addresses, so some might not consider it a(n explicit)
memory address computation - there is some nuance there.
It's one place that someone might not read/understand/implement the spec
correctly.
Which is my superpower.

closed 2023-04-13 by jinyinghan324