riscv/riscv-isa-manual #1010: Do LR and SC instructions which trap invalidate the active reservation set?

JamesKenneyImperas, 2023-04-12:
Section 10.2 (Load-Reserved/Store-Conditional Instructions) does not seem to describe the effect of LR and SC instructions that do not complete (because they trap) on any active reservation set. Do these instructions invalidate the current reservation set when they trap, or not?

Thanks.

aswaterman, 2023-04-12:
The spec indicates the situations that cause the reservation to be yielded and doesn't mention this case, so yielding the reservation is not required.

However, as an entirely separate matter, the reservation can be yielded at any time for any reason, provided eventual forward progress isn't impeded.  This implies that an implementation could yield the reservation on exceptions if it wanted to.

JamesKenneyImperas, 2023-04-13:
Ok, this is implementation-defined. Should the specification be updated to indicate that either behaviour is legal for these instructions if they trap? Although this is covered by the catch-all that the reservation may be invalidated at any time for any reason, these two instructions are special because they explicitly modify the reservation set in normal operation.

On a related issue, I could only find a statement that executing `LR` invalidates any active reservation set in this non-normative note:

> The invalidation of a hart’s reservation when it executes an `LR` or `SC` imply that a hart can only hold one reservation at a time...

I think this should be stated in the normative text (if it isn't already and I've missed it). This note is also the cause of my confusion that prompted this issue: I wasn't sure whether an `LR` or `SC` that traps counts as "executing" or not.

Thanks.

aswaterman, 2023-04-13:
Re: the second issue, I believe the relevant normative fact is that an SC can only pair with the most recent LR in program order (hence any older reservation sets aren't relevant).

Timmmm, 2024-03-25:
The spec is very unclear on this. It says

> Regardless of success or failure, executing an SC.W instruction invalidates any reservation held by this hart.

which really *sounds* like the instruction will always invalidate the reservation. However from what you guys are saying (and the Sail model) it is only required to cancel the reservation if it retires successfully?

Definitely needs clarification in the spec.

gfavor, 2024-03-25:
It sounds like the last post is about speculative execution (distinct from SC success/failure).

Architecturally an SC either is performed (whether it succeeds or fails) or is not performed (e.g. a trap of some sort is instead taken, possibly as a result of attempting to execute the instruction).  The spec says what is required when the instruction is performed, and covers cases of of not being performed by allowing (within constraints to ensure eventual success) a reservation to be invalidated for unexpected/random/non-deterministic reasons - which can include attempted but aborted speculative excecution of an SC.

A good implementation will not invalidate the reservation if an SC is aborted.  But it is certainly allowable within the constraints of ensuring eventual success.

Given that SAIL can't model all the possible valid implementations, it makes sense that SAIL models the clean and probably typical behavior that an SC invalidates the reservation only when it is performed.

Timmmm, 2024-03-26:
Sounds logical - I just think the difference between executing but failing an SC and executing but trapping needs to be explicit. It's quite vague at the moment. Something like:

> Regardless of success or failure, an SC.W instruction that retires MUST invalidate any reservation held by this hart. An SC.W instruction that does not retire due to a trap is not required to invalidate the reservation.

(Off topic but I really wish the spec used the standard unambiguous MUST/MAY/SHOULD system - see #1144)

allenjbaum, 2024-03-26:
I think the vagueness hinges on the use of succeed +fail. That is whether
or not the instruction writes the destination or not.
An SC that aborts doesn't fail, because it never completed, never wrote the
destination, so it isn't required to invalidate a reservation.
Reservations can be invalidated by events that have nothing to do with an
access to the reservation set addresses, and those can be implementation
dependent,
so my interpretation is that an implementation "may" invalidate a
reservation if it takes an exception, but isn't required to, and may not
even be recommended to ("should").
Taking an exception might always trigger a timeout, so the end result would
be the same, of course.