riscv/riscv-isa-manual #1011: How should design process unimplemented pmp csr and performance counter？

jinyinghan324, 2023-04-13:
If 16 PMP CSRs are implemeted, how should design process csr instructions to pmpaddr16~63? Illegal instruction or read-only 0？

And if 20 performance counter(mhpmcounter3-22) are implemented, how should design process csr instructions to mhpmcounter23-31？ Illegal instruction or read-only 0？

scottj97, 2023-04-13:
"Implemented", when it comes to CSRs, means that they do not take an illegal instruction exception. Read-only 0 counts as "implemented". For PMP, when the spec says "Implementations may implement zero, 16, or 64 PMP entries" that refers only to which ones take an illegal instruction exception. You can make any number of the implemented PMP CSRs read-only 0.

For your specific question about pmpaddr16-63, you can do either illegal instruction or read-only 0, but they must all have the same behavior. (For example, having pmpaddr16-31 read-only 0 and pmpaddr32-64 take an illegal instruction exception would not be a legal implementation.)

For performance counters, the spec says, "All counters should be implemented, but a legal implementation is to make both the counter and its corresponding event selector be read-only 0." Which means you must not take an illegal instruction exception on any of the 29 event counters, but mhpmcounter23-31 and mhpmevent23-31 can be read-only 0.

pdonahue-ventana, 2023-04-13:
Adding emphasis:
> For performance counters, the spec says, "All counters **should** be implemented, but a legal implementation is to make both the counter and its corresponding event selector be read-only 0." Which [...]

I think that you **should** not take an illegal instruction exception on any of the counters, though that is technically allowed.  (I encourage you to not make an implementation that tests the limits of the architecture.  I'm just being pedantic.)

closed 2023-04-14 by jinyinghan324