riscv/riscv-isa-manual #1016: No new PDF 'releases' generated since end of Jan

asb, 2023-04-18:
Prior to Jan 31st, it seems PDFs were generated very regularly (every commit?). But as of last now, the last release of a draft PDF is https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20230131-c0b298a

Did something break with the automation?

rpsene, 2023-04-18:
@aswaterman I was told you create the releases manually for the main branch.

aswaterman, 2023-04-18:
@rpsene Nope.  We rely on github actions to do that, but as Alex said, it has been broken since January.

rpsene, 2023-04-18:
@aswaterman let me raise a PR with a fix.

rpsene, 2023-04-18:
@asb fix in place https://github.com/riscv/riscv-isa-manual/releases/tag/isa-449cd0c

asb, 2023-04-19:
Thanks! Marking this issue closed as it appears to be be fixed.

closed 2023-04-19 by asb