riscv/riscv-isa-manual #1019: Converting riscv-isa-manuals asciidoc results in text-less wavedrom diagrams because of 3rd party bug

damageboy, 2023-04-22:
I realize the the asciidoc versions of the isa-manual is not complete, and I'm mostly reaching out through this issue to @wmat in hopes there is some secret knowledge about wavedrom-cli that could be shared.

In short, I'm trying to generate the asciidoc html/pdf versions locally after replicating the build environment but I'm getting an annoying artifact in both the pdf and the html versions where the wavedrom diagrams are generated without any text rendered inside them:

[image]

Trying to render this directly through wavedrom-cli replicated the bug when rendering to PNG only and not with SVG,
and seems to be a [known](https://github.com/wavedrom/cli/issues/22) issue reported in the wavedrom repo and [unfixed in node-svg2img](https://github.com/fuzhenn/node-svg2img/issues/41).

Is there a change that a PR for switching to generating SVGs instead of PNGs would be accepted for the asciidoc version given this (unresolved) bug and SVGs generally being slightly more friendty (IMO) for zooming in / printing / hidpi in general?

damageboy, 2023-04-22:
Just in case it wasn't entirely clear I'm effectively suggesting to submit a PR of all the modified wavedrom files after running:

```
sed -i 's/^\[wavedrom, ,\]/[wavedrom, ,svg]/' src/images/wavedrom/*.adoc
```

Unfortunately I did not find an easy way to perform this change globally through a theme file or otherwise, although my asciidoctor is still very shallow.

wmat, 2023-04-24:
Unfortunately I don't have any secret knowledge of wavedrom-cli to share.
If you're having issues building locally, I'd suggest trying the docker
based build environment found here:
https://github.com/riscv/riscv-docs-base-container-image

As for switching to all svg imaging, I'm fine with that.  In fact, I've
been moving more toward using bytefield-svg for diagramming as it allows
for more flexibility.

damageboy, 2023-04-27:
I'll submit the change to use svg in that case, while not everything is converted bytefield-svg

wmat, 2023-04-27:
It just occurred to me that you may be missing some fonts.  WavedRom diagrams require the Helvetica font, I believe.  Also, I build locally and it works for me.

closed 2023-04-27 by wmat

wmat, 2023-05-15:
Note that this change renders all wavedroms without any color, effectively ignoring the "type" attribute.  I'm currently looking for a fix.

PR #1025 (2023-04-27): switch all wavedrom images to output svg
fixes #1019 1019