riscv/riscv-isa-manual #1024: Build instructions

timsifive, 2023-04-27:
I know how to build the LaTeX doc:
```
cd build
make
```

Which is somewhat intuitive. How do I build the asciidoc document? It would be nice to have a section in README.md that describes both.

wmat, 2023-04-27:
That hasn't changed, Tim.  What has changed is that typing make will build
all versions of both docs, i.e. unpriv and priv, both html and pdf.
Results will still be in the build directory.

I'll add some better instructions to the README.

timsifive, 2023-04-27:
Then where does the regular isa manual end up? After `make` in build/, I
just have:
```
Makefile              riscv-privileged.bbl  riscv-privileged.log
 riscv-privileged.pdf
riscv-privileged.aux  riscv-privileged.blg  riscv-privileged.out
 riscv-privileged.toc
```

Tim

wmat, 2023-04-27:
Did you update everything after the branch name change?

git branch -m riscv-isa-asciidoc main
git fetch origin
git branch -u origin/main main
git remote set-head origin -a

timsifive, 2023-04-27:
> Did you update everything after the branch name change?

Looks like I didn't `git pull` another time, but now I think I'm up to date.

> git branch -m riscv-isa-asciidoc main
> git fetch origin
> git branch -u origin/main main
> git remote set-head origin -a

I shouldn't need this, do I? I'm just on `main`,
change 31ed9c3c18d6ec9e98aebd58c1cbaa41da48ae66, which matches the latest
on https://github.com/riscv/riscv-isa-manual.

Looks like something is broken in my asciidoc install, and I didn't see any
error until I explicitly requested `make unpriv`. After fixing all that,
now it works. Thanks. (Related: It would be more helpful to me if
https://github.com/riscv/docs-dev-guide/blob/main/local_build.md just
contained a list of all the gems and npm things that needed to be
installed, rather than working through them one by one.)

Tim

wmat, 2023-04-27:
>
> > Did you update everything after the branch name change?
> >
>
> Looks like I didn't `git pull` another time, but now I think I'm up to
> date.

Great! Glad you're up to date.

>
>
> > git branch -m riscv-isa-asciidoc main
> > git fetch origin
> > git branch -u origin/main main
> > git remote set-head origin -a

Agreed.  I figured I'd throw it out there in case you were still on the old
branch.

> I shouldn't need this, do I? I'm just on `main`,
> change 31ed9c3c18d6ec9e98aebd58c1cbaa41da48ae66, which matches the latest
> on https://github.com/riscv/riscv-isa-manual.
>
> Looks like something is [...]

Noted.  I'll add that to the docs-dev-guide.

closed 2023-10-11 by timsifive