riscv/riscv-isa-manual #1026: Clarification around VS interrupts

jiahzhang, 2023-04-27:
Is it allowed for VS interrupts to take effect when virtualization is off? spike seems to indicate [this](https://github.com/riscv-software-src/riscv-isa-sim/blob/fcbdbe7946079650d0e656fa3d353e3f652d471f/riscv/processor.cc#L662), but I don't see it in the hypervisor spec anywhere.

scottj97, 2023-04-27:
```
    if (state.v && enabled_interrupts == 0) {
```

If not in VS or VU mode, `state.v` will be false, so VS-level interrupts will not be taken.

Isn't that what the spec requires?

jiahzhang, 2023-04-27:
Yes, I'm wondering where the spec says this?

scottj97, 2023-04-27:
Lots of discussion about this a few years back:
* https://github.com/riscv/riscv-isa-manual/issues/633
* https://github.com/riscv/riscv-isa-manual/pull/634
* https://github.com/riscv/riscv-isa-manual/commit/bf971f91f95e9c9cece7f9a137b3e276a66332f5

The current text that specifies this is in the Hypervisor Privilege Modes section:

> HS-mode is more privileged than VS-mode, and VS-mode is more privileged than VU-mode. VS-mode interrupts are globally disabled when executing in U-mode.

And under the description of `mstatus.MIE`:

> When a hart is executing in privilege mode _x_, interrupts are globally enabled when xIE=1 and globally disabled when xIE=0. Interrupts for lower-privilege modes, w<x, are always globally disabled regardless of the setting of any global wIE bit for the lower-privilege mode.

The combination of those two statements means that when the privilege level is M, HS, or U, interrupts destined for VS-level are globally disabled.

jiahzhang, 2023-04-27:
Great, thanks for clarifying!

scottj97, 2023-04-27:
Please close the issue if your question is resolved.

closed 2023-04-27 by jiahzhang