riscv/riscv-isa-manual #1027: FCVTMOD.W.D Instruction in Zfa

liangkaiwang, 2023-04-28:
@aswaterman 

I have a question regarding to the instruction, **Modular Convert-to-Integer Instruction**.

- What Javascript function does this instruction emulate? 
- In one of the paragraph, it states
`Bits 31:0 are taken from the rounded, unbounded two's complement result, then sign-extended to XLEN bits and written to integer register {\em rd}. $\pm\infty$ and NaN are converted to zero.` 

If my understanding of the above is correct, this instruction converts the double precision input into 2's complement signed number with unlimited precision (on both the integer and fraction parts). Since the rounding mode is RTZ, if the intermediate 2's complement signed number is a negative number, we actually need to "increment" on the integer part to get the rounded integer. In other words, we can have a case when the input is a very small negative value, it can be rounded "up" to positive zero. Are we sure we really want RTZ instead of RNI?

And for the next question. After the rounded integer is obtained, we then take bit 31 to 0 of the rounded result with infinite "integer" precision. So what this means is, a negative number may become positive, and vice versa. Are we sure this is the expected behavior

aswaterman, 2023-04-28:
I can understand why you'd look askance at the semantics, but the answer to both of your questions is: yes, that was deliberate.  ARMv8's similar instruction, FJCVTZS, does indeed round towards zero.  And, in the event of overflow, it also discards the MSBs, even if doing so changes the sign of the result.

cc @gfavor @ptomsich, who had more of a hand in the definition of this instruction than me.

gfavor, 2023-04-28:
On Thu, Apr 27, 2023 at 11:14 PM Andrew Waterman ***@***.***>
wrote:

> I can understand why you'd look askance at the semantics, but the answer
> to both of your questions is: yes, that was deliberate. ARMv8's similar
> instruction, FJCVTZS, does indeed round towards zero. [...]

All I would add on is simply that the semantics of this and ARM's similar
instruction are driven by the desired semantics for efficiently supporting
JavaScript's convert operation.  This link provides some more what and why
info about this instruction and also answers your question:
https://stackoverflow.com/questions/50966676/why-do-arm-chips-have-an-instruction-with-javascript-in-the-name-fjcvtzs

Greg

ptomsich, 2023-04-28:
Please refer to the ECMAscript (a.k.a. JavaScript) standard for semantics required in JavaScript:
- https://262.ecma-international.org/13.0/#sec-toint32
- https://262.ecma-international.org/13.0/#sec-touint32

For the To(U)?Int{16,8} semantics, the decision was to not provide a dedicated instruction and rather defer to a FCVTMOD.W.D and a subsequent extension operation.

--Philipp.

liangkaiwang, 2023-04-28:
Thank you all for the prompt response. Let me dig into this to gain some
background knowledge. This is an "interesting" instruction and I would love
to know how many people depend on that feature when the input is going
out-of-bound.

Liang-Kai

aswaterman, 2023-04-28:
Yeah, it’s certainly an odd duck.

liangkaiwang, 2023-05-01:
Just wonder, I think this instruction was proposed as a scalar instruction as of now. What's the odd of this getting added into the Vector Extension?

ptomsich, 2023-05-01:
@liangkaiwang Could you provide some background on the use-cases for a SIMD version of this instruction?

liangkaiwang, 2023-05-01:
I actually don't have one. But given what I have seen so far, it appears to me that several of the arithmetic instructions end up getting promoted into the Vector ISA and that's why I wonder if there has been push by someone else to get this into the Vector ISA.

nick-knight, 2023-05-01:
If we're sharing our vector Zfa Christmas lists, I want vector versions of `fround.*`, for use in vector transcendentals.

kasanovic, 2023-05-02:
@nick-knight Please start new issues for new features.

kasanovic, 2023-05-02:
@liangkaiwang This is instruction is intended for use by the Javascript JIT engine, which is unlikely to ever do auto-vectorization.  If that ever happens, we could be motivated to add a vector variant.

aswaterman, 2023-05-02:
Agreed on both counts.

Closing this issue, as I believe it's resolved, but please feel free to reopen if need be.

closed 2023-05-02 by aswaterman