riscv/riscv-isa-manual #1031: [Question] fflags for vector floating point non-arithmetic instructions

liangkaiwang, 2023-05-03:
Hi,

In section 13.1, I see

_13.1. Vector Floating-Point Exception Flags 
A vector floating-point exception at any active floating-point element sets the standard FP exception flags in the fflags register. Inactive elements do not set FP exception flags._ 

But does this apply to the instruction that only moves the data from one element to another like vfmerge, vfmv, vfslide1[up/down], etc? As an example, if the input is Signaling NaN, do we raise the invalid flag for the instruction? 

thanks,
Liang-Kai

aswaterman, 2023-05-03:
Moves don't give rise to floating-point operations, in this sense of the term, and so they don't set any FP exception flags.

liangkaiwang, 2023-05-03:
Thanks for your quick response. Really appreciate it. 

Given that there are some non-intuitive behavior surrounding NaN Boxing and NaN canonicalization, I feel it may be worthwhile to specify them in the spec. But I will leave it up to the committee to decide if it is useful.

If I want to summarize the behavior, for data-move instructions like vfmerge, vfmv, vfslide1up, and vfslide1down, we have
- NaN boxing check needs to be applied when the input is a floating point scalar
- After NaN Boxing check is applied, the data is treated as a bag of bits and data is populated to the vector elements based on the instructions behavior and the elements' status without any modification
- For those data-move instructions, exception flags are never raised under any condition

thanks,
Liang-Kai

aswaterman, 2023-05-03:
Your summary sounds right.  I'll leave it to the authors of the relevant specs to decide if clarification is needed.  (IIRC, it's sufficiently specified, but the information is scattered around.)