riscv/riscv-isa-manual #1034: Is mip.SSIP level or transition sensitive?

jrahmeh, 2023-05-05:
Section 3.1.9 of the privileged spec says:
```
If supervisor mode is implemented, bits mip.SSIP and mie.SSIE are the interrupt-pending and interrupt-enable bits for supervisor-level software interrupts. SSIP is writable in mip and may also be set to 1 by a platform-specific interrupt controller.
(edited)
```
If the the SSIP bit is settable to 1 by a platform-specific controller, does it get set whenever the controller bit is 1, or whenever it transitions from 0 to 1, or is it up to the implementation?

zhuotianshu, 2024-02-26:
I also have the same question. And I also wonder why SSIP is not lile SEIP which is read as the logical-OR of interrupt controller signal and the software writeable bit:

> Supervisor-level external interrupts are made pending based on the logical-OR of the softwarewritable SEIP bit and the signal from the external interrupt controller.

zhuotianshu, 2024-02-26:
And, can SSIP also be clear by an interrupt controller?

gfavor, 2024-02-27:
To the first question (from last year), it is up to the "platform-specific controller" as to when it sets SSIP to 1.

An interrupt controller cannot clear SSIP.  Per the spec it can only set SSIP to a 1.  In essence this setting of SSIP is an "edge" that sets SSIP to 1 in that moment (and doesn't keep forcing SSIP to a 1).  Which I believe matches the model of software sending a software interrupt to a hart as a form of doorbell (and not as a level signal that remains asserted and needs to be cleared at some later point in time based on some sort of hardware feedback to the sender).

In contrast, external interrupts are "level" signals that remain asserted until software (usually an interrupt handler) tells the hardware interrupt controller that it can now deassert the interrupt request signal.

Also note that there is a natural OR'ing of multiple software interrupts being sent to a hart since they all just keep setting SSIP (which is what you want for a "doorbell" interrupt mechanism).  And there isn't the hardware concern that one has with external interrupts that one wants to keep the external interrupt request signal asserted until all sources of the interrupt have been serviced.  For software interrupts the handling model is a little different.  (A better answer that talks about how external interrupts and software interrupts typically work in a system is more detailed, but not one I have the time to write at length about.  Hopefully this provides a rough idea for why SSIP was architected the way it is.)

zhuotianshu, 2024-02-27:
> To the first question (from last year), it is up to the "platform-specific controller" as to when it sets SSIP to 1.
>
> An interrupt controller cannot clear SSIP. Per the spec it can only set SSIP to [...]

Understood. Thank you very much!

zengderui, 2024-09-14:
Hi,
As per the comments, SSI is edge-triggered and MIP.MSIP is writable.  What about MSI? The privileged spec says that MIP.MSIP is not writable, does this mean MSI is level-triggered?

aswaterman, 2025-12-09:
The spec says that an interrupt is taken whenever the corresponding bits in `mip` and `mie` are both set (and some other stuff), implying that `mip` and `mie` are sampled continuously.  This is consistent with a level-sensitive interrupt model, though there's obviously an isomorphism.