riscv/riscv-isa-manual #1035: Inconsistency between AsciiDoc version and LaTeX version of unpriv spec

ZenithalHourlyRate, 2023-05-05:
It seems that the main branch switched to a AsciiDoc version of unpriv spec and it is released. However, after some rough inspection (when writing [this](https://github.com/chipsalliance/rocket-chip/pull/3329#discussion_r1186267239)) of the new release I found that some changes of the LaTeX version were not in the AsciiDoc version, for example,

https://github.com/riscv/riscv-isa-manual/commit/f5f9c27010b69a015958ffebe1ac5a34f8776dff
https://github.com/riscv/riscv-isa-manual/commit/db7a4a0dad0e99d1ec1fc67b582624fc0aeae98e

are not in https://github.com/riscv/riscv-isa-manual/blob/63aeaada9b2fee7ca15e5c6b6a28f3b710fb7e58/src/naming.adoc

Are there any measures taken to ensure that the AsciiDoc version contains all the changes of the LaTeX version? Or should we "backport" these commits one by one?

gfavor, 2023-05-05:
> Yes, prior to releasing the asciidoc version, a line by line comparison
> was done between the asciidoc version and the latex version. It would seem
> that this change may have slipped in after that comparison was completed.
> I'm happy to make these changes to the asciidoc version, or would you
> prefer to issue a pull request?

Bill,

How do we ascertain what all Latex changes occurred after the compare point
(so that all of them get applied to the new asciidoc version)?

Greg

ZenithalHourlyRate, 2023-05-05:
> would you prefer to issue a pull request?

#1036. Seems that a lot of changes were not applied.

gfavor, 2023-05-05:
Bill, do we know how this situation happened?  Not just for the sake of
making sure we catch all the missing changes, but also so that we can
recognize what we need to do wrt the Priv conversion so as to avoid this
from happening there.

wmat, 2023-05-05:
Greg, I had been monitoring the LaTeX master branch for changes and manually applying them to the asciidoc branch prior to switching over.  Note that the master latex branch was renamed to the latex branch on which the last commit was April 18th.  Any new changes to LaTeX after that date should only be to content in the Privileged spec.

I'm not sure how the change was missed, as I'd normally see the commit by Andrew in my inbox.  I do not have a commit email for these changes for some reason.  Going forward I'll monitor GitHub daily instead of relying on email notifications.

gfavor, 2023-05-05:
Thanks.  It sounds like we'll be ok for Priv with your extra monitoring
(and Andrew will knowingly coordinate with you).  Fyi for others, Andrew
and I at this point are holding off making any Priv changes until we can do
them on top of a reasonably clean asciidoc version of the Priv spec that
will hopefully be available very soon.

So, for Unpriv, we just need to make sure that all these recent Unpriv
changes are identified and applied to the asciidoc version.  Is there a
belief that #1036 <https://github.com/riscv/riscv-isa-manual/pull/1036> covers
all the missing changes?

Message ID: ***@***.***>
>

ZenithalHourlyRate, 2023-05-05:
> Is there a belief that #1036 covers all the missing changes?

No. I only checked naming.tex/adoc.

wmat, 2023-05-05:
Closing with merge of #1036 . Will audit LaTeX commits for any other missed commits.

closed 2023-05-05 by wmat

aswaterman, 2023-05-05:
Following up on what @gfavor wrote, this won't happen again going forward, since all future work will happen exclusively on the AsciiDoc source.  But I agree, @wmat, that a manual auditing of the last few months' commits is advisable.