riscv/riscv-isa-manual #1043: Sv57x4 upper 3 bits of virtual address

jrahmeh, 2023-05-16:
The virtual address in Sv57x4 is 59-bits wide and corresponds to the guest physical address.  But the guest physical address is only 56 bits wide.  Are bits 56, 57, and 58 of the virtual address always zero?

scottj97, 2023-05-16:
Why do you say it's 56 bits wide? See figure 109 in the latest priv-isa-asciidoc.pdf. The guest physical in Sv57x4 is 59 bits.

I guess you are assuming the guest physical comes from a VS-mode translation (vsatp.MODE=Sv39, Sv48, or Sv57), which does restrict it to 56 bits (so 3 upper bits of 0), but that is not necessarily so. If vsatp.MODE=Bare then the guest physical could use the full 59 bits.

jrahmeh, 2023-05-16:
First stage translation with sv57 will translate a 57-bit guest virtual address to a 56-bit guest physical address.
Second stage sv57x4 translation requires a 59-bit address as input.  We have a 56-bit output from 1st stage and a 59-bit input to second stage.  What to do with the missing 3 bits? Hardwire to zero?

scottj97, 2023-05-16:
I don't see any other reasonable option, so yes.

closed 2023-05-16 by jrahmeh