riscv/riscv-isa-manual #1048: encodings for LQ/SQ

tariqkurd-repo, 2023-05-22:
Hi,

I can't see LQ/SQ in the encodings table, either in an older Latex version or the newer Asciidoc version of the unpriv spec.
I can find this text:
```
A LDU (load double unsigned) instruction is added using the existing LOAD major opcode, along
with new LQ and SQ instructions to load and store quadword values. SQ is added to the STORE
major opcode, while LQ is added to the MISC-MEM major opcode.
```
but no definition. Have they been allocated?

I can find the 16-bit encodings but not the 32-bit ones.

Tariq

tariqkurd-repo, 2023-05-22:
I also have a case of `cbo.inval` dissassembling as `lq` so there seems to be a problem here.

kasanovic, 2023-05-22:
There is no design for RV128 at this point,  never mind a frozen version, so LQ/SQ have not been allocated .  The RV128 chapter should be considered as just an early concept draft.

aswaterman, 2023-05-23:
@tariqkurd-repo That might be a bug in whichever disassembler you're using.  RV128 opcodes haven't been formally allocated, so disassemblers shouldn't be attempting to match them.

tariqkurd-repo, 2023-05-23:
OK - so RV128I is listed as 1.7 draft, but I thought that 1.0 meant frozen or ratified. Presumably the numbering scheme has changed since that number was assigned. if it's just an early draft I'd say 0.1 is appropriate as a version.

It's also confusing that encodings for LQ/SQ aren't listed but C.LQ, C.LQSP, C.SQ, C.SQSP are.

The reason that this has come to light is that in RISC-V CHERI they use the XLEN*2 load/store encoding for their capability (fat pointer) type. For RV32 it's fine, they repurpose LD/SD. For RV64 it's not fine as LQ/SQ don't have real encodings.

Allocating an encoding to SQ shouldn't be a problem as there are several funct3 values available in the STORE opcode group, but for LQ AFAIK only funct3=7 is available in the LOAD opcode group..

Unfortuantely they are currently using funct3=2 and MISC_MEM which clashes with the cache management ops CBO.*

So my question is - is it possible to allocate encodings to LQ/SQ? Not just for CHERI but also for a recently discussed proposal for permitting XLEN*2 load/store. This would also benefit from having encodings allocated.

What's your take on this @kasanovic , @aswaterman 

Thanks, Tariq

aswaterman, 2023-05-23:
I don't view this as a priority because we're talking about a base ISA that hasn't been defined yet and [what's currently] a custom extension.  Neither of those things needs a standard opcode assignment at this time.

But, FWIW, my current suggestion, which shouldn't be taken as binding, is that LOAD funct3=7 is the right place to put LDU, STORE funct3=4 is the right place to put SQ, and MISC-MEM with funct3=4 is the right place to put LQ (since its funct3 field matches SQ).

tariqkurd-repo, 2023-05-24:
ok, thanks. That's good enough for now.