riscv/riscv-isa-manual #1049: Evaluation of interrupt in case of CSR access

amorenosmd, 2023-05-26:
Following the RISC-V spec section 3.1.9:

> An interrupt i will trap to M-mode (causing the privilege mode to change to M-mode) if all of
the following are true: (a) either the current privilege mode is M and the MIE bit in the mstatus
register is set, or the current privilege mode has less privilege than M-mode; (b) bit i is set in both
mip and mie; and (c) if register mideleg exists, bit i is not set in mideleg.

> These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time
from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated
immediately following the execution of an xRET instruction or an explicit write to a CSR on which
these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).

When an interrupt becomes visible in mip, the spec indicates that the interrupt doesn’t need to be immediately taken. Instead, there may be a small delay before evaluation, except for the case that an xRET instruction is executed. Additionally, performing an explicit CSR write on certain registers also cause the interrupt to be taken immediately after execution. This raises a couple questions:

- Is every CSR access instruction a CSR write? For example, is a “csrrs rd, csr, x0” considered a write in this context, even though it’s not really writing anything?
- In case of write, the trap must be taken immediately after executing such instructions. Imagine executing a CSR read/write of the mip register which is already updated with an interrupt that must be taken (i.e. conditions (a), (b) and (c) are all fulfilled for that interrupt). We could load the mip value, which already contains the interrupt bit set, into a scalar register before taking the interrupt. Is this compliant with the spec?

gfavor, 2023-05-27:
Table 11.1 in the latest Unpriv spec defines which CSR instructions do and don't write the CSR.

The second excerpted paragraph answers the second bullet question, i.e. immediately **after** execution of a CSR write instruction (which means after the instruction has updated its dest reg with a value indicating the pending interrupt), the pending interrupt must be taken.

amorenosmd, 2023-05-29:
Thanks for the answer @gfavor. Just one more question to make sure I'm interpreting this correctly:

It is then compliant with the spec to read the MIP register while it contains an interrupt that must be taken, yet still execute and retire a few instructions after reading the MIP register before taking the trap?

gfavor, 2023-05-29:
Yes, except for the exception cases.

The Priv spec's MIP description answers this question when it says "These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated immediately following the execution ..."

amorenosmd, 2023-05-30:
All clear then. Thanks for the answers @gfavor.

closed 2023-05-30 by amorenosmd