riscv/riscv-isa-manual #1057: Are the hardware performance counters hart-local CSRs?

henry-hsieh, 2023-06-30:
In the Zicnt spec, the CSRs are given with exact behaviors. For example, the `cycle` CSR should return core-wide cycle count (not by hart). The `time` CSR should return system-wide real-time clock. The `instret` CSR should return hart-local instruction retired count.

In the Zihpm spec, the hardware performance counters are not given these constraints. In the SMT design, is it legal that the harts in the same core share the same hardware performance counters? Could a hardware performance counter shared across the system for system performance monitoring? Or it's just implementation defined?

allenjbaum, 2023-07-01:
All CSRs are in theory hart private, that is, each hart has its own copy
and only that hart can access it.
The MCYCLE CSR is explicitly an exception:

The mcycle CSR may be shared between harts on the same core, in which case
writes to mcycle will be visible to those harts. The platform should
provide a mechanism to indicate which harts share an mcycle CSR.

No other exceptions are listed in the spec.
Having said that, there is no  prohibition that I can see about whether events
that are counted by an HPM counter must be hart local.
Events are defined by the platform, so a highly threaded core could have an
HPM counter event that counts events outside the hart.
E.g.  all the cache misses on the core, summed across all harts on that
core.
That still doesn't allow the CSR to be shared, as far as I can tell, so if
one hart clears its HPM, no other hart would.

I am not a definitive source of knowledge, however, as I've discovered on
more than one occasion......

henry-hsieh, 2023-07-01:
Hi Allen,

Thanks for the explanation. After studying the privileged and unprivileged spec, I thought that almost everyone will agree your claim:
> All CSRs are in theory hart private, that is, each hart has its own copy and only that hart can access it.

The CSRs should be used to control and provide the status of the hart.

However, the Zicsr extension says:
> 1. RISC-V defines a separate address space of 4096 Control and Status registers associated with each hart.
> 2. Most CSRs (including, e.g., the fcsr) are not visible to other harts

The privileged spec only covers CSR addresses and fields. It makes me wonder which CSRs can be visible to other harts. The cycle definitely should be one. It's legal that each hart has separate address space of CSRs but some of them mapping to the same physical register.

In my opinion, the spec should make more clear about the constraint of CSRs. Maybe add following sentence while introducing the CSRs:
> Unless otherwise specified, the CSRs are not visible to other harts by default.

ptomsich, 2023-07-02:
Henry,

On Sat 1. Jul 2023 at 18:05, Henry Hsieh ***@***.***> wrote:

> Hi Allen,
>
> Thanks for the explanation. After studying the privileged and unprivileged
> spec, I thought that almost everyone will agree your claim:
>
> All CSRs are in theory hart private, that is, each hart has its own copy
> and only that hart can access it.
>
> The CSRs should be used to control and provide the status of the hart.
>
> However, the Zicsr extension says:
>
>
>    1. RISC-V defines a separate address space of 4096 Control and Status
>    registers associated with each hart.
>    2. Most CSRs (including, e.g., the fcsr) are not visible to other harts
>
> The privileged spec only covers CSR addresses and fields. It makes me
> wonder which CSRs can be visible to other harts. The cycle definitely
> should be one. It's legal that each hart has separate address space of CSRs
> but some of them mapping to the same physical register.
>
> In my opinion, the spec should make more clear about the constraint of
> CSRs. Maybe add following sentence while introducing the CSRs:
>
> Unless otherwise specified, the CSRs are not visible to other harts by
> default.
>
> The specification would be clearer by applying the word “visible” to “CSR
contents” rather than “CSRs”.

Conceptually, you have only hart-private CSRs.
However, the contents of some CSRs may offer view at the overall system
resource (e.g., a synchronized cycle counter — whether the implementation
actively replicates the values each cycle or retrieves it from a single
external counter register during readout) and therefore the same CSR
contents may be visible through the CSRs of each hart.  The access is
always done through the hart-private CSR though (as there is no possibility
to read/write the CSR of a different hart directly).

Hope that clarifies the specification language.

Thanks,
Philipp.

allenjbaum, 2023-07-02:
If writing a CSR on one hart cannot affect the value seen by reading the
CSR from a different hart,
then they are really completely separate CSRs, even if the same eternal
event can affect them.
That's how I interpret the cycle counter; every cycle on that core will be
counted by each hart separately,
but writing one has no effect on any other.
That means that an HPM counter could even count some event on some other
hart (.which is a security issue..)

henry-hsieh, 2023-07-03:
Allen's definition on private CSRs is more close to what I thought. The issue would be whether the CSRs can only be hart-private. If the CSRs can only be hart-private, it would be illegal that implementation has custom core control register such as hardware thread control, shared cache control, etc. in an SMT core. However, I agree that the standard CSRs are hart-private and have no exception.

allenjbaum, 2023-07-03:
Yes, for custom...it's custom, and you could share. You could also make the
register be MMIO accessible instead of CSR op accessible.

henry-hsieh, 2023-07-04:
Thanks for sharing your opinions! I'll close the issue.

closed 2023-07-04 by henry-hsieh