riscv/riscv-isa-manual #1060: When the core is in Machine mode, and MXLEN is 64 and UXLEN is 32, what value will be obtained when using the CSR instruction to read INSTRET?

stephen-ji, 2023-07-09:
From the SPEC, we can see that the width of INSTRET is determined by UXLEN. When the core is in Machine mode, and MXLEN is 64 and UXLEN is 32, what value will be obtained when using the CSR instruction to read INSTRET? Should it be the low 32 bits of INSTRET (sign-extended to 64 bits) or the original 64-bit INSTRET (the high 32bits is INSTRETH)? 
When the core is in Machine mode, and MXLEN is 64 and UXLEN is 32, what value will be obtained when using the CSR instruction to read INSTRETH?

gfavor, 2023-07-09:
See Priv section 2.6 CSR Width Modulation.

stephen-ji, 2023-07-09:
> See Priv section 2.6 CSR Width Modulation.

Thanks for your reply. But my question is not about the modification of CSR width. I will use another example to describe my question:
The layouts of SSTATUS are different when SXLEN=64 and SXLEN=32. When the core is in Machine mode, and MXLEN is 64 and SXLEN is 32, which layout will be used when using the CSR instruction to read SSTATUS.

gfavor, 2023-07-09:
Since sstatus is defined to have width SXLEN (versus simply XLEN - which is always the current XLEN), sstatus always has that width even when read from a mode operating with a different XLEN (such as in your example).  So when M-mode reads sstatus with MXLEN=64 while SXLEN=32, I believe CSR width modulation applies to convert the current 32-bit sstatus value into a 64-bit value returned by the CSR instruction.

stephen-ji, 2023-07-10:
> Since sstatus is defined to have width SXLEN (versus simply XLEN - which is always the current XLEN), sstatus always has that width even when read from a mode operating with a different XLEN (such [...]

Got it. Thanks!

closed 2023-07-10 by stephen-ji