riscv/riscv-isa-manual #1068: Clarification of HINT instructions

jaimiejellema, 2023-07-17:
In section 2.9, about HINTS it is stated that:
> Implementations are always allowed to ignore the encoded hints.
To me, that means that an implementation which recognizes all defined HINTS, but chooses not to act on them, would still be a conforming implementation. Furthermore this appears to imply that a conforming implementation could decide on a per HINT basis, for each time it executes, if it want to act on it.

However the draft CFI defines behavior, that upon detection that the HINT bits match the bits in x7, and raise an illegal instruction exception if that is not the case. In my opinion that behavior implies that my above presumption no longer hold, since an implementation HAS to act on  the encoded HINTS. A supported extension might be active or in-active but I feel like deciding on a per HINT basis is distinctly different from activating or deactivating an extension, on presumably a larger block of code.

I see 3 resolutions:
1) The wording of the HINT is clarified to state that implementations that recognize the hint must act on it, but extensions might make exceptions. The later to allow freedom in for example branching hints.
2) The wording of the HINT is clarified to state that extensions might redefine this behavior and/or restrict acceptable behavior of conforming implementations

aswaterman, 2023-07-17:
The CFI spec is essentially designing a new ISA mode that is consciously backwards incompatible.  Under that premise, the "fix" is for the CFI spec to redefine these opcodes to not be HINTs anymore when the new mode is active.  This preserves the definition of HINT.  cc @ved-rivos -- I guess the discussion can move over to https://github.com/riscv/riscv-cfi if need be.

ved-rivos, 2023-07-18:
The CFI spec addresses this as follows:

>      The `lpad` instruction uses the code points defined as HINTs for the `AUIPC` opcode. When 
>      Zicfilp is not active at a privilege level or when the extension is not implemented, the landing
>      pad instruction executes as a no-op. A program that is built with `lpad` instruction can thus
>      continue to operate correctly, but without forward-edge control-flow integrity, on processors
>      that do not support the Zicfilp extension or if the Zicfilp extension is not active.

@jaimiejellema - please let me know if this text sufficiently addresses this.

jaimiejellema, 2023-07-19:
Yes, thank you that answers my questions.

closed 2023-07-19 by jaimiejellema