riscv/riscv-isa-manual #1070: When csr menvcfg.PBMTE=0,  whether Svpbmt extension is not available in S-mode but can be available in U-mode?

zhangjingru111, 2023-07-18:


gfavor, 2023-07-18:
Svpbmt is either completely enabled or completely disabled by menvcfg.PBMTE.  When enabled, it affects address translation of all addresses - whether S-mode or U-mode addresses.

zhangjingru111, 2023-07-19:
> 

"The PBMTE bit controls whether the Svpbmt extension is available for use in S-mode and G-stage
address translation (i.e., for page tables pointed to by satp or hgatp). When PBMTE=1, Svpbmt
is available for S-mode and G-stage address translation. When PBMTE=0, the implementation
behaves as though Svpbmt were not implemented."
That's what the spec says. It doesn't mention U mode. So, does PBMTE also affect U mode by default?

gfavor, 2023-07-19:
Svpbmt affects address translation - which applies to virtual addresses in
both S-mode and U-mode.  And if Svpbmt is disabled, then that is true for
address translation in both modes as well.

Greg

Greg

closed 2023-07-20 by zhangjingru111