riscv/riscv-isa-manual #1080: [question/idea] Indexed CSR access?!

stnolting, 2023-08-05:
Hey!

Maybe this is the wrong place for ISA proposals (especially when coming from a random community enthusiast 😅) and maybe my idea is way too specific and would be better off as a custom extension.... But anyway, I'm curious what you think about this. 

## Why?

I often see (and actually use) low-level machine-mode code where CSR accesses are abstracted by some function like this:

```c
uint32_t csrrw(int address, uint32_t write_data); // return <= CSR[address] <= wdata
```

Here, `address` could be an actual variable that is not static at compile-time, which is quite handy for handling array-like CSRs (e.g. PMP or HPM). However, implementing such a function normally results in large switch/case constructs increasing code size and slowing down execution speed.

So I was looking through the RISC-V ISA space to see if there is something that would provide a more ~elegant~ _efficient_ way to implement this. I found the [RISC-V Advanced Interrupt Architecture (AIA)](https://github.com/riscv/riscv-aia) extension proposal and - as a "spin-off" from that - the [riscv-indirect-csr-access](https://github.com/riscv/riscv-indirect-csr-access) extension proposal. 

However, as far as I understand these two ISA extensions go the long way as they're using "indirect" indexed addressing: two specific CSRs are added. The first one is used to specify the CSR address and the second one for actually accessing the addressed CSR. So the indexed access requires two consecutive instructions. Furthermore, in AIA the address range seems to be constrained to some extension-specific CSRs only.

Why not have a dedicated instruction for a "more directly" indexed CSR access that uses a general purpose integer register to index an arbitrary CSR being accessed?

## How?

One could add a new instruction "CSRRWX" (CSR read-write-inde**X**ed) that reads and writes a CSR based on an address derived from the source register rs2. The same CSR address-, privilege- and read/write-checks as in `Zicsr` would be used. From a hardware point of view this is quite easy to implement for the data path: just a 12-bit 2-to-1 mux selecting either the instruction's immediate value or rs2.

I crawled trough the instruction encodings to find out that in the SYSTEM opcode space `Zicsr` uses 6 out of 8 `funct3` encodings:``

[image]

`funct3 = 000` is used by the other SYSTEM instructions, but `100` sees to be unused yet. So here is my actual proposal:

## `Zicsrx` - X-Register-Indexed CSR Access

[image]

```
csrrwx rd, rs2, rs1 : rd <= CSR[rs2[11:0]] <= rs1
```

(Another two bits from `funct7` could be used to also implement read-set and read-clear instructions.)

sorear, 2023-08-07:
This is a perennial proposal, see for instance [this isa-dev thread from 2018](https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/gGXSjYcY-74/m/qLhjkrmvAAAJ).

The implementation is more complicated than you make it sound because different CSRs are accessed at different points in the pipeline, and an indirect access must assume the worst case of any CSR (waiting for previous floating point instructions for fflags, blocking future floating point for frm, blocking future loads and stores for mstatus, blocking future instruction fetches for mpmpcfg, etc); indirect CSR access as defined in the AIA spec only supports a class of CSRs that have restricted synchronous effects on instruction execution.  Realistically, csrrwx would have to wait for all previous instructions and block all subsequent instructions, and would be slower in most cases on most implementations than a well-predicted switch statement.

On a more surface level, your proposed encoding steps on the same minor opcode used for Zicfiss MOP instructions, and this repository hasn't been the correct place to start an extension proposal for years; you want lists.riscv.org.

stnolting, 2023-08-12:
Thanks for the feedback and for the link!

closed 2023-08-12 by stnolting