riscv/riscv-isa-manual #1086: Version of V in isa manual

christian-herber-nxp, 2023-08-08:
> "V" Standard Extension for Vector Operations, Version 1.0

> This document is version 1.1-draft of the RISC-V vector extension.

> This version holds updates gathered after the start of the public review. The spec will have a final update to version 2.0 at time of ratification.

@wmat If the draft with latest changes was incorporated, it is clearly not 1.0. If this version is supposed to be called 1.1-draft, this would indicate it will become a 1.1 version without post-fix later. A version 2.0 would indicate incompatibility with version 1.0, and thats hopefully not the case.

wmat, 2023-08-11:
I simply integrated what is in the Vector repository, so I leave the answering of this question to those familiar with the Vector spec versioning. @aswaterman can you comment?

aswaterman, 2023-08-11:
I’ll leave it to the V TG to resolve this.