riscv/riscv-isa-manual #1100: Confusion about OPMP NA4/NAPOT  programming when G>2

tariqkurd-repo, 2023-08-22:
This issue: https://github.com/riscv/riscv-isa-manual/issues/933 touches on the topic but doesn't cover everything I'm concerned about.

In my example the PMP granularity `G>2`

The ISA manual says:

Two other address-matching modes are supported: naturally aligned power-of-2 regions (NAPOT), *including the special case* of naturally aligned four-byte regions (NA4)

When G >= 1, the NA4 mode is *not selectable*. When G >= 2 and pmpcfgi.A[1] is set, i.e. *the mode is NAPOT*,  then bits pmpaddri[G-2:0] read as all ones.

A is a WARL field - so in my mind NA4 is an illegal programming, but _pmpcfgi.A[1]_ above destroys this theory. the terms _not selectable_ and _special case_ just ocnfuse the issue further.

So can someone tell me what happens when G>2 and A=NA4?
It doesn't help that the SAIL model doesn't support G, so this behaviour isn't modelled, so there is no golden reference behaviour (unless I should be looking in Spike?)

aswaterman, 2023-08-22:
It’s already well defined. A is a WARL field, so trying to write NA4 to A will not succeed. Following the usual WARL rules, A could become any other legal value on such an attempt. (On implementations I’ve worked on, A becomes NAPOT in this case.)

tariqkurd-repo, 2023-08-23:
it is certainly true that if NA4 is an illegal value then A[1] is enough to detect NAPOT - but that's a simple logic optimisation so I wonder why it's in the spec. My reading of it was that either NA4/NAPOT could be programmed (so bit 0 is don't care) which is why I doubted that NA4 was an illegal programming. I then wondered if the WARL behaviour of the A field was only referring to it being read-only-zero when the PMP region isn't implemented but the CSRs are present.

This confusion was compounded by the SAIL model which doesn't treat NA4 as an illegal value if G>2 and has a couple of comments referring to similar confusion, and doesn't treat A as WARL at all:

```
    NA4   => { // TODO: I find the spec unclear for entries marked NA4 and G = 1.
               // (for G >= 2, it is the same as NAPOT). In particular, it affects
               // whether pmpaddr[0] is always read as 0.
               let lo = pmpaddr << 2;
               Some((lo, lo + 4))
```

If the spec had said `When G >= 2 and pmpcfgi.A[1]` had been `When G >= 2 and pmpcfgi.A=NAPOT` I thinkthat I would have reached the intended conclusion.

aswaterman, 2023-08-23:
The current text is precise, though I agree it is terse.  I don't think there's anything to fix here.

aswaterman, 2023-08-23:
BTW, Spike correctly supports G>0, so you can use that as a reference in the meantime.

tariqkurd-repo, 2023-08-23:
There are many such cases in the RISC-V ISA manual, and other architectures are no different to be fair, it's not a unique problem. Now I know the answer, the text makes sense, but before I knew the answer the text was ambiguous.

Removing the reference to A[1] would have avoided the confusion. I'll leave it at that.

closed 2023-08-23 by tariqkurd-repo