riscv/riscv-isa-manual #1102: Question about U-Mode XLEN=32 and M/S-Mode XLEN=64 CSR and instruction behavior

fanghuaqi, 2023-08-29:
Hello, according to this post here https://riscv.org/blog/2023/01/run-32-bit-applications-on-64-bit-linux-kernel-liu-zhiwei-guo-ren-t-head-division-of-alibaba-cloud/ , user can modify U-Mode XLEN from 64 to 32 in a RV64 CPU to make 32bit application able to run in RV64 CPU in user mode.

Here I have questions below:

1. What is the behaviors when access `cycleh/timeh/instreth` in U-Mode XLEN=32, when M/S Mode XLEN=64, raise an illegal instruction exception, or return the corresponding higher 32bits of the registers
2. What is the behavious of instructions such as `rdtimeh/rdcycleh/c.fsw/c.flw` in U-Mode  XLEN=32, when M/S Mode XLEN=64, raise an illegal instruction exception, or execute it as normal.

Thanks

gfavor, 2023-08-29:
On Tue, Aug 29, 2023 at 2:58 AM Huaqi Fang ***@***.***> wrote:

> Hello, according to this post here
> https://riscv.org/blog/2023/01/run-32-bit-applications-on-64-bit-linux-kernel-liu-zhiwei-guo-ren-t-head-division-of-alibaba-cloud/
> , user can modify U-Mode XLEN from 64 to 32 in a RV64 CPU to make 32bit
> application able to run in RV64 CPU in user mode.
>
>    1. What is the behaviors when access cycleh/timeh/instreth in U-Mode
>    XLEN=32, when M/S Mode XLEN=64, raise an illegal instruction exception, or
>    return the corresponding higher 32bits of the registers
>
> U-mode is architecturally oblivious to what lies above it in S-mode and
M-mode, including whether SXLEN is different than UXLEN.  So if the
Counters extensions are implemented, then these *h CSRs must also be
implemented.

>
>    1. What is the behavious of instructions such as
>    rdtimeh/rdcycleh/c.fsw/c.flw in U-Mode XLEN=32, when M/S Mode XLEN=64,
>    raise an illegal instruction exception, or execute it as normal.
>
> Ditto.

The general idea is that U-mode architecture and its XLEN is not affected
by S-mode's XLEN.  It is S-mode software's responsibility to deal with a
U-mode process running under it that has a different XLEN (i.e. XLEN=32).
For example, dealing with ECALLs from an XLEN=32 U-mode process.

Greg

Message ID: ***@***.***>
>

fanghuaqi, 2023-08-30:
Hi @gfavor , thanks for your reply.

According to ISA spec here, this `RDTIMEH` only present when XLEN=32, so does it mean in U Mode, XLEN=32, it will present, but in S Mode XLEN=64, it will not present.

[image]

Similar cases for `rv32fc` only instructions `c.flw` present here,
[image]

What confuse me is when the U-Mode XLEN is writable, if this CPU CORE is RV64, and M/S XLEN=64, and U Mode XLEN changed from XLEN=64 to XLEN=32, then the CPU CORE XLEN is 64 or 32? And according to spec, some instructions and csrs only exist in RV32, so I am not sure when U Mode XLEN changed from XLEN 64 to 32, then the non-existing instructions and csrs should be present now or not?

Thanks

gfavor, 2023-08-30:
On Tue, Aug 29, 2023 at 6:45 PM Huaqi Fang ***@***.***> wrote:

> According to ISA spec here, this RDTIMEH only present when XLEN=32, so
> does it mean in U Mode, XLEN=32, it will present, but in S Mode XLEN=64, it
> will not present.
>
The way to generally think about changing XLEN is that the privilege mode
in question sees and only sees what would be RV32 architecture if XLEN=32
or RV64 architecture if XLEN=64 (i.e. instructions, CSRs,
instruction behaviors, etc).  Software running in an RV32 mode, for
example, will see no more and no less that what is defined for RV32.  This
then implies that CSRs and instructions that are specific to RV64 must
appear to be absent in RV32, and vice-versa.  For your RDTIME example, in
RV32 software sees 32-bit RDTIME and RDTIMEH CSRs, and in RV64 software
sees a 64-bit RDTIME CSR.

Greg

Message ID: ***@***.***>
>

fanghuaqi, 2023-08-30:
OK, thanks @gfavor , then it is clear to me.

aswaterman, 2025-12-09:
Closing as answered.