riscv/riscv-isa-manual #1104: What exactly "takes effect immediately" upon changing satp.MODE?

dramforever, 2023-08-31:
https://github.com/riscv/riscv-isa-manual/blob/ee240a4e33ab4d64f300274532a3afa6e33dbd51/src/supervisor.adoc#L978-L982

It's unclear what exactly "takes effect immediately" and "without the need" for `sfence.vma`. In particular, it's unclear whether writing `satp.MODE` has any effect on the translation cache.

Say, if I change `satp.MODE` from `Bare` to some of `Sv*`, is it guaranteed that on the very next instruction to be fetched, the `pc` is translated using the page table and and will not use the cached `Bare` translation?

dramforever, 2023-08-31:
I guess I'm mostly confused as to whether "without the need to execute an SFENCE.VMA instruction" means it has the effect of `sfence.vma` or some other meaning.

The address translation thing is just all very special in the privileged architecture and never follows normal rules of ordering, so it would be great if the guaranteed effects are spelt out more clearly.

aswaterman, 2023-08-31:
Here, “takes effect immediately” means that the very next instruction fetch uses the new translation mode. It does not imply that it has the effect of an SFENCE.VMA.

dramforever, 2023-08-31:
I still think it's unclear that if the implementation caches `satp.MODE = Bare` identity translations, they won't be used for the next instruction fetch. It seems that only a matching `ASID` is required, not a matching `MODE`.

https://github.com/riscv/riscv-isa-manual/blob/ee240a4e33ab4d64f300274532a3afa6e33dbd51/src/supervisor.adoc?plain=1#L1329-L1332

(Caching  `satp.MODE = Bare` identity translations appears to be explicitly allowed)

https://github.com/riscv/riscv-isa-manual/blob/ee240a4e33ab4d64f300274532a3afa6e33dbd51/src/machine.adoc?plain=1#L3064-L3069

dramforever, 2023-09-01:
Having went through it again, I think it's clear (even though slightly counterintuitive) that conceptually any translation cache would be caching PTEs, not mappings, so changing `satp.MODE` in this case would immediately modify memory access to (conceptually) use the new translation algorithm with no cached PTEs (because there was no `satp.MODE` to begin with).

Thanks for the explanations.

closed 2023-09-01 by dramforever