riscv/riscv-isa-manual #1116: Accessing an unimplemented CSR is reserved, not necessarily illegal

aswaterman, 2023-09-01:
The old clause was incorrect, as sometimes a virtual instruction exception must be raised instead.

(It's also inconsistent with the rest of RISC-V, which recommends but does not mandate exceptions for reserved encodings.  And, furthermore, the existence of non-conforming extensions made the old mandate toothless: an implementation could purport to have a non-conforming extension at every otherwise-unused CSR address.)

ved-rivos, 2023-09-07:
I understand that the requirement for triggering an "illegal instruction" pertains only to non-existent i.e. to unimplemented CSRs. 

If so, then non-conforming extensions would be exempt from this rule because they do end up **implementing** the CSR; hence, they aren't bound by this stipulation as the CSR exists. Only a CSR **implemented** either as a standard or a non-conforming extension would sometimes cause a "virtual instruction" exception as it exists.. 

If access to non-existent i.e. an **unimplemented** CSR could exhibit unspecified behavior, it might be beneficial to recommend a range of safe behaviors such as treating it as a no-op. At the very least, it may be recommended that the unspecified behaviors should not result in undesirable consequences like privilege escalation, system hangs, deadlocks, or other forms of denial-of-service.

aswaterman, 2023-09-07:
The problem is that an implementation can simply claim that it has no unimplemented CSRs: it has a non-conforming extension that covers the entirety of the CSR space that isn't used by the standard extensions it implements.  So any requirement imposed on unimplemented CSRs can be trivially circumvented.

ved-rivos, 2023-09-07:
What was confusing me is that if the implementation has no unimplemented CSRs i.e. every CSR has been implemented through a non-confirming or standard or custom implementation then it would not be bound by this requirement since this pertains to only **unimplemented** CSRs. Such implementation that has implemented every CSR would likely provide a definition of the implementation of each CSR to its software and each such CSRs would have an implementation defined behavior. The architecture could still provides a defined behavior for **unimplemented** CSRs; else each implementation may need to reason what to do about them. It may be okay to say behavior of unimplemented CSRs is unspecified, if that is the intent, but is recommended to be otherwise safe and secure.

pdonahue-ventana, 2023-09-08:
> I think I'd like a better definition of "implemented every CSR".

I think that his point was that even if you're required to trap on unimplemented CSRs, some implementation can still decide not to trap and still be legal.  They implement the N CSRs defined in the priv spec and they claim that the other 4096-N CSR numbers are custom non-conforming CSRs that _are_ implemented.  One of those may alias to one of the N implemented ones and this behavior is a "nonconforming extension" even though it's really just lazy decode.  Another may be read-only-zero which is also a custom nonconforming extension.

One way of doing things is to not require a trap on unimplemented CSRs (which is the proposal).  Another way is to require traps and then say that if you don't trap then that behavior is a nonconforming extension.  They're both equivalent in practice.  I personally prefer the latter since it introduces a bias toward trapping and on the surface it appears more similar to other architectures (and will be more familiar and comfortable for newcomers).

allenjbaum, 2023-09-08:
I don't think I disagree - but in the case someone has asserted that they
*don't* implement some extension that defines a CSR,
yet lists that CSR as implemented, then ACTs (or something) can catch that
non-conforming case - if we want to catch that case.

To do that, the ACT for a "trap-on-unimplemented CSRs" extension just tests
all CSRs that are
undefined by extensions that are claimed to be supported by the
implementation (in the non-custom space only)
regardless of what the riscv-config file says. (The other ACTs will already
test the CSRs for extension claimed to be supported)

ved-rivos, 2023-09-08:
I'm not chiefly concerned with implementations that might sidestep ACT by claiming to implement a non-existent CSR. Such concerns fall more under ACT's domain and likely don't represent a pervasive architectural issue.

It's entirely fitting for reserved CSRs — those designated for future standard purposes — to be repurposed for non-conforming extensions. The crux of the matter is discerning the behavior of non-existent CSRs. This refers to those that haven't been activated for non-conforming or custom extensions and CSRs linked to standard extensions unsupported by the given implementation.

Should an attempt to access a CSR not result in a trap, it implies the existence of an underlying mechanism. This could be something as basic as advancing the program counter (`pc`), effectively making it a no-op. Given this, the behavior of a CSR should invariably align with one of the following criteria: a) Undertake an action, even if it's as minimal as a no-op, thus confirming its presence. b) Trigger a trap, marking its absence.

Ideally, the implementation won't craft CSRs as aliases in potentially hazardous ways. An example of such unsafe behavior might involve aliasing CSR number 005  directly to 305h. However, if such an approach is taken by an implementation, it shouldn't be expressly prohibited, as it remains within the realm of a non-conforming extension.

pdonahue-ventana, 2023-09-08:
> in the case someone has asserted that they
*don't* implement some extension that defines a CSR,
yet lists that CSR as implemented, then ACTs (or something) can catch that
non-conforming case - if we want to catch that case.

Does anyone assert that they _don't_ implement an extension or do they just list the extensions that they _do_ implement?

The point of nonconforming extensions is that they're using standard encodings for custom purposes.  If I don't implement the H extension then I'm free to have a nonconforming extension that uses the hedeleg CSR encoding for my own purposes.  ACTs shouldn't fail if hedeleg is used by a nonconforming extension (unless H is claimed to be supported).  That defeats the whole purpose of nonconforming extensions.

> Ideally, the implementation won't craft CSRs as aliases in potentially hazardous ways. An example of such unsafe behavior might involve aliasing CSR number 005 directly to 305h.

I also thought of that, but I think that the RISC-V philosophy is that if you want to be stupid then that's your right.  I just hope that downstream customers know about any such stupidity.

ved-rivos, 2023-09-08:
> I also thought of that, but I think that the RISC-V philosophy is that if you want to be stupid then that's your right. I just hope that downstream customers know about any such stupidity.

That is what I suggested the sentence following the quoted sentence.

pdonahue-ventana, 2023-09-08:
> That is what I suggested by the sentence following the quoted sentence.

Sorry, I didn't mean to imply otherwise.  I was agreeing with you and then saying that I hope that customers are fully aware of anything stupid.

allenjbaum, 2023-09-08:
I think I agree with all of the above.
There is a good point that if there is an extension that has defined CSRs,
and an implementation does not claim to support it,
those should be available for reuse as custom (non-conforming) extension
use.
That does change the details of what I expressed above, but not the spirit
;
I think it means we should only be checking CSRs that are undefined by any
ratified spec (for a profile),
not those that are undefined for the extensions that are claimed to be
supported.

closed 2024-01-30 by aswaterman

cebarobot, 2024-05-15:
> The problem is that an implementation can simply claim that it has no unimplemented CSRs: it has a non-conforming extension that covers the entirety of the CSR space that isn't used by the standard extensions it implements. So any requirement imposed on unimplemented CSRs can be trivially circumvented.
> 
> A recommendation that reserved encodings raise exceptions is fine (but it has no normative impact). IIRC some such statement already exists in the unprivileged spec.

@aswaterman Sorry to bother, but I didn't find such statement in the unprivileged spec. Could you please point it out for me?

In fact, I think the ISA spec should give the implementation some guide about how to deal with accessing a non-existent CSR.

cebarobot, 2024-05-15:
> You didn’t find such a statement since it doesn’t exist. It follows from a logical progression of statements that do exist in the spec.

Ok, I got it. Thanks a lot.

commit 14367227 (2024-01-29): Attempts to access non-existent CSRs are reserved

Resolves #1116
commit 516685da (2024-01-29): Attempts to access non-existent CSRs are reserved

Resolves #1116
PR #1207 (2024-01-30): Attempts to access non-existent CSRs are reserved
Resolves #1116
commit 246efe08 (2024-01-30): Attempts to access non-existent CSRs are reserved (#1207)

Resolves #1116