riscv/riscv-isa-manual #1120: Question about the purpose of xscratch CSR

KevinKBytes, 2023-09-11:
Hello, according to RISCV-ISA, purpose of mscratch is as follows : "Typically, it is used to hold a pointer to a machine-mode hart-local context space and swapped with a user register upon entry to an M-mode trap handler".

My question is what about nested exceptions? Woudn't it swap back to the original user register value upon entry to second exception? Or is there a flag where software can query to know that it is about to enter a nested exception/trap, and skip swapping mscratch value out?

aswaterman, 2023-09-11:
In the Linux kernel, the invariant is that `sscratch` holds a nonzero value when executing in user-space and holds zero when executing in the kernel.  So the nested case can be detected and handled separately by branching on the zeroness of `sscratch`.

There's obviously a transient state where the invariant doesn't hold, but interrupts are always disabled when in the transient state, and the code is written such that a synchronous exception can't occur.

https://github.com/torvalds/linux/blob/0bb80ecc33a8fb5a682236443c1e740d5c917d1d/arch/riscv/kernel/entry.S#L17

KevinKBytes, 2023-09-12:
"Code is written such that a synchronous exception can't occur."

By this do you mean the Linux kernel exception handlers themselves are guaranteed to not generate exceptions, hence there is no need to support nested exceptions?

aswaterman, 2023-09-12:
That requirement applies only to the window where sscratch is in a transient state—-roughly speaking, until the context has been spilled to memory. I suggest reading the code.