riscv/riscv-isa-manual #1128: Add RAS local interrupt fields to {m,s,vs}{ie,ip} CSRs

ved-rivos, 2023-09-13:
@jhauser-us @aswaterman 
Is the AIA spec integration imminent into Priv 1.13? If not should we introduce the interrupt numbers 35 and 43 defined by AIA spec as local interrupts for low-priority and high-priority RAS events as a temporary measure?

aswaterman, 2023-09-13:
cc @gfavor

gfavor, 2023-09-13:
I'd say yes, let's go ahead and incorporate the interrupt #35/#43 definitions from AIA into the Priv spec since it will be a while before AIA gets properly integrated (i.e. next year).

jhauser-us, 2023-09-18:
Without extension Smaia/Ssaia, the Priv spec says that interrupt numbers 16 and higher are "designated for platform or custom use", meaning the Priv spec doesn't determine what they mean.  How then can the Priv spec (without Smaia/Ssaia) provide definitions for interrupt numbers 35 and 43 that it expressly doesn't control?

Ved wrote:
> If not should we introduce the interrupt numbers 35 and 43 defined by AIA spec as local interrupts for low-priority and high-priority RAS events as a temporary measure?

Temporary measure for what purpose?  For people who don't implement Smaia/Ssaia?  I'm going to go out on a limb and suggest that people who do implement Smaia/Ssaia are probably aware of the AIA document.

gfavor, 2023-09-18:
I would take back my last post and instead agree with John.  Even when AIA is "properly integrated", the Priv spec should continue to say what it says and it is the AIA extensions that gives specific meaning to interrupt numbers 35 and 43.

Put the other way around, John argues that as even other bits among bits 16+ are defined over time by other extensions, the Priv spec should not change and each other extension will document the specific meaning it introduces around specific interrupt numbers.

aswaterman, 2023-09-18:
OK to close this issue, then, Ved?

ved-rivos, 2023-09-18:
I was hoping to get some clarifications to the Priv spec. 

The Priv spec states "Bits 15:0 are allocated to standard interrupt causes only, while bits 16 and above are designated for platform or custom use.". Is "platform use" synonymous with "custom use"?  Even if not it is still stated as designated for custom use. The Smaia/Ssaia defines codes 16-12 and 32-47 as reserved for standard extensions. 

This may seem like a contradiction to the Non-standard encoding space guarantee provided otherwise. Should this change in designation reflect in Priv. specification (retroactively) to avoid the contradiction?

aswaterman, 2023-10-31:
Platform use means the hardware platform defines the usage.  I know that sounds circular, but the point is, it isn't synonymous with custom, since custom uses might conflict with platform uses.

Platforms might be standard, in which case they may allocate some of these interrupts for standard use; a correct implementation of that platform cannot reallocate those interrupts for custom uses.  However, a standard platform is free to designate a subset of the platform-use interrupts as custom.

(Platforms might also be custom, but then the intended behavior is obvious.)

@gfavor and I concluded the best resolution is to delete the words "or custom".  This doesn't prevent those interrupts being used for custom purposes, but it makes it clear that the platform gets to make the decision.

commit ea578803 (2023-10-31): Clarify meaning of "platform or custom use"

Resolves #1128
closed 2023-10-31 by aswaterman