riscv/riscv-isa-manual #1132: Zcmop and depending compressed extension

a4lg, 2023-09-19:
It theoretically requires existence of the 16-bit encoding space (and possibly the smallest compressed instruction subset, `Zca`).

However, just requiring `Zca` causes problems (recent Toolchain SIG discussion suggests separating `C` and its subset `Zca` as two "bases" and not implying `Zca` from `C`).

The most realistic solution (as of now) would be to write like:

> `Zcmop` requires either `C` or `Zca` extension.

and let the toolchain to handle the conflicts.  The toolchain will not be able to imply anything from `Zcmop` but able to check whether `Zcmop` and either `C` or `Zca` co-exists (note that, it means that we have to specify both `zcmop` and base compressed base extension in the architectural string like `-march=rv64gc_zimop_zcmop` or `-march=rv32e_zce_zcmop`).

aswaterman, 2023-09-20:
Not sure I agree.  We're converging elsewhere on M implying Zmmul https://github.com/riscv/riscv-isa-manual/pull/1121 -- I realize this is a pain for the toolchain devs, but the toolchain devs aren't the only constituency here, and in any case they can't unilaterally define this policy.  I'll leave this open until the M/Zmmul issue is decided; don't expect a decision until then.

nick-knight, 2023-09-20:
I'm not familiar with Zcmop. Do its encodings or CSRs overlap with other extensions? If not, it may be sufficient to simply say that it relaxes the lower bound on ILEN to 16 (or whatever the C chapter says).

aswaterman, 2023-09-20:
I also considered that approach, @nick-knight, and I do think that would suffice. But I’d rather force the issue re: Zca and C. If things go the way you’ve proposed re: Zmmul/M, then Zca/C will fall in line.

aswaterman, 2023-09-20:
I moved the M->Zmmul scheme forward.  Let's assume the same will happen for C->Zca.  Then, I think this issue devolves to Zcmop->Zca.

aswaterman, 2023-09-20:
@tariqkurd-repo following the chaos in #1121, do you agree that the Zca spec should be updated to indicate that C implies Zca?  If so, can you make it so?

tariqkurd-repo, 2023-09-24:
That sounds ok to me, I'm out for the next week so I'll try and do it when I'm back.

a4lg, 2023-10-21:
Since `C` now implies `Zca`, unless we need a good reason to depend directly on `C`, depending on `Zca` would be better.

> `Zcmop` requires the `Zca` extension.

aswaterman, 2023-10-24:
@a4lg Indeed.

commit b7c3a8e9 (2023-10-24): Zcmop requires Zca

Resolves #1132
closed 2023-10-24 by aswaterman