riscv/riscv-isa-manual #1134: Change of  modes:  Machine mode/usermode to supervisor mode in riscv using spike simulator

shruthihk, 2023-09-21:
I am trying to change the privilege level  i.e from machine mode to supervisor mode tried with below things.
Whenever the trap comes the mstatus is getting updated i.e  mpp bit is setting to 11(machine mode) so we tried to write the mpp bit to 10(supervisor mode)  and tried checking the sstatus and scause register, these register are not updating. Even tried with delegation register by setting stip/stie  bit it's also not working . Please let us know how to fix this

allenjbaum, 2023-09-21:
I, for one, am unclear how you think you are changing modes.
The MMP field is a status field, not a control field - writing it to 10 has
no effect.
The MPP field is only used by the MRET instruction which uses it to return
to the mode encoding in MPP
Likewise, changing MEPC doesn't change the PC of the running code,
until you execute an MRET.
Please read priv spec section 3.1.6.1

shruthihk, 2023-09-22:
Hi,
Please let me know how to change the modes inn riscv using spike simulator

allenjbaum, 2023-09-22:
I. don't understand the question. There is nothing special - no
extra support, in Spike to change modes.
You use MRET to go to a lower prive mode, assuming you've set all the right
CSR parameters correctly,
and use ECALL or EBREAK to go to a higher priv mode, also assuming you've
set up all the right CSR parameters correctly.
The right parameters is homework for you.

shruthihk, 2023-09-22:
Thanks for the response, We are working on it .