riscv/riscv-isa-manual #1143: Unimplemented PMPs

Timmmm, 2023-10-09:
The spec says:

> Up to 64 PMP entries are supported. Implementations may implement zero, 16, or 64 PMP entries; the
lowest-numbered PMP entries must be implemented first. All PMP CSR fields are WARL and may be
read-only zero.

This seems odd to me. Are you allowed to support e.g. 4 PMP entries by implementing 16 PMP registers but having 12 of them be read-only zero? If so why restrict the number of implemented PMP entries to 0, 16, 64?

A similar part of the spec is with HPM counters, but there the spec says that all HPM counters "should" be implemented, but can be read-only zero. Why not do the same for PMP entries?

allenjbaum, 2023-10-09:
A fine question... which may become moot.
Yes, if any PMP entries are implemented, then all 16 must be.
Yes, they are allowed to be readOnly zero - but most importantly, must not
trap.
IF you implement 16, and try to read the 17th, then it will trap (at least
now) unless all 64 are implemented.
It would be legal to claim all 64 are implemented, even if only 4 are
writable and the rest readOnly zero (or, actually, some other readOnly
value, I believe).

I think the spec is being updated so that it will be legal to not trap if a
CSR isn't implemented
(much like executing an undefined op is not guaranteed to trap with an
illegal op exception)
 though a profile could mandate that accesses to unimplemented CSR will
trap.

Omarezz123, 2024-04-16:
Spec not mention what if no pmp is implemented what will happen when accessing them will give 
illgeal read or make them read_zero (so not give illegal read)

gfavor, 2024-04-17:
The answer is the same as for the CSRs of any extension that is not implemented.  See chapter 2 of the **_latest_** integrated adoc version of the Priv spec.

Omarezz123, 2024-04-18:
can i have a link ?

gfavor, 2024-04-18:
[https://github.com/riscv/riscv-isa-manual](url)

nick-knight, 2024-04-18:
@Omarezz123 
1. Go here: https://github.com/riscv/riscv-isa-manual/releases (I have this in my browser bookmark toolbar.)
2. Identify the latest release: it should be closest to top of page.
3. Click "Assets": Volume I of the ISA specification is titled "unpriv-isa-asciidoc.pdf", and Volume II is "priv-isa-asciidoc.pdf". Volume II is the one Greg is pointing you to.

closed 2025-12-09 by aswaterman