riscv/riscv-isa-manual #1144: Spec does not define "should"

Timmmm, 2023-10-09:
Most formal specs I've read define words like "should", "must", "may", up front, but the RISC-V spec does not appear to. Despite that "should" appears dozens of times, often in places where it looks like they really meant "must", e.g.

> In systems without S-mode, the medeleg and mideleg registers should not exist.

> An Exception Code of 0 is reserved to mean "unknown cause" and implementations that do not distinguish sources of NMIs via the mcause register should return 0 in the Exception Code.

I think it would be helpful to audit those and probably change them to *must*.

Ideally adopt [RFC2119](https://datatracker.ietf.org/doc/html/rfc2119) definitions and use them.

ethindp, 2023-12-02:
It might be worth adopting RFC 2119 plus the RFC 8174                                            updates, requiring that those words appear in capitals to have the intended effect.

radimkrcmar, 2025-04-15:
I fully support this idea.  It does require a lot of care when rewriting the manual, though:

> In systems without S-mode, the medeleg and mideleg registers should not exist.

This probably should stay a "should", because a related note explains that the choice is there because of backward compatibility.  I don't think we should say that medeleg and mideleg must not exist without S -- it's closing design options without a good reason.  Leaving them unspecified seems better in this case.

> An Exception Code of 0 is reserved to mean "unknown cause" and implementations that do not distinguish sources of NMIs via the mcause register should return 0 in the Exception Code.

This also seems fine as "should", because mcause on NMI is unspecified (implementation-defined) except for MSB, and changing it is again possibly invalidating existing implementations.

This thread was recently brought up in a discussion about another should:

> SRET must be provided if supervisor mode is supported, and should raise an illegal-instruction exception otherwise.

I think this should is correct, because it's referring to the unspecified behavior when decoding a reserved instruction.

> SRET should also raise an illegal-instruction exception when TSR=1 in mstatus, as described in Section 3.1.6.6.

This "should" is even murkier and almost looks like a "must", but SRET must not raise the exception when called from M-mode, even if mstatus.TSR=1.

aswaterman, 2025-12-09:
Resolving this will require a careful audit, but we "should" do so.