riscv/riscv-isa-manual #1147: state enable bit for small priv-1.13-era extensions

aswaterman, 2023-10-10:
Going forward, we'd like to allocate only a single stateen bit for small extensions defined between priv-spec releases, rather than allocating a bit for every small extension.  This isn't an absolute rule; it's just an attempt to cut down on stateen bit count when practical.

The plan is that the qosid register will be controlled by the priv-1.13 stateen bit, and possibly other registers will, too.

Note this isn't an absolute policy; it's just an attempt to cut down on stateen bits when practical.

ved-rivos, 2023-12-01:
@aswaterman is there a position defined for this bit.

aswaterman, 2023-12-02:
I'm pretty swamped right now. Can you make a proposal (and a PR once we an agree)?

ved-rivos, 2023-12-02:
Will do. 

Bits that are presently defined in `xstateen0`.
- Bit  0 - Custom State
- Bit  1 - `fcsr` for Zfinx and related extensions
- Bit  2 - `jvt` CSR for Zcmt
- Bit 56 - CD ?
- Bit 57 - `hcontext` CSR
- Bit 58 - all IMSIC state, including CSRs `stopei` and `vstopei`
- Bit 59 - all other state added by the AIA and not controlled by bits 60 and 58
- Bit 60 - CSRs `siselect`, `sireg`, `vsiselect`, and `vsireg`
- Bit 63 - controls access to `sstateen0` and `hstateen0`

This issue proposes to define a single `xstateen0` bit - bit 55 - to govern new state introduced by Priv-1.14-era extensions.
- Bit 55 - all state introduced by extensions included in Privileged ISA version 1.14

The state of the following extensions is expected to be governed by the bit 55:
- Ssqosid - CSR `srmcfg` (holds QoS identifiers)
- Ssclic - CSRs `stvt`, `snxti`, `sintstatus`, `sintthresh`, `sscratchcsw`, and `sscratchcswl` (S-mode CLIC extensions)
- Ssctr - indirect state accessed by `siselect`/`vsiselect` value of 0x200 through 0x2FF (Control transfer records extension)

ved-rivos, 2023-12-14:
Bit 55 in CSR `mstateen0` controls access to CSR `henvcfgh`.

Bit 56 in CSR `xstateen0` is defined to control access to new state introduced by Priv-1.14-era extensions as follows:
- CSR `srmcfg` introduced by the forthcoming Ssqosid extension.