riscv/riscv-isa-manual #1150: Question about execution order between lw/sw from different Hart

fennecJ, 2023-10-14:
Hi, I'm currently watching Write subsumption example of RVWMO:
[image]
| Hart 0 | Hart 1 | 
| -------- | -------- | 
| li t1, 3     | li t3, 2| 
| li t2, 1     | | 
| (a) sw t1,0(s0)| (d) lw a0,0(s1) | 
| (b) fence w,w | (e) sw a0,0(s0) | 
| (c) sw t2,0(s1)| (f) sw t3,0(s0) | 

And the doc says:
• (a) precedes (c) in the global memory order because of rule 2
• (c) precedes (d) in the global memory order because of the Load Value axiom
• (d) precedes (e) in the global memory order because of rule 7
• (e) precedes (f) in the global memory order because of rule 1

My question is, how can we ensure (c) precedes (d) because of Load Value axiom.
The definition of Load Value axiom appears at 「17.1.4. Memory Model Axioms」:
Each byte of each load i returns the value written to that byte by the store that is the latest in global
memory order among the following stores:
1. Stores that write that byte and that precede i in the global memory order
2. Stores that write that byte and that precede i in program order

Based on my understandings, the program order means execution order of instructions for each hart.
For the Write subsumption example above, the program order of Hart0 will be (a), (b), (c);
and for Hart1 the program order will be (d), (e), (f).

If Hart 0 and Hart 1 have their own write buffer,
and the result of (c) haven't drain to main memory and appear in global memory order, 
may Hart1 not aware of write operation by Hart0's (c) and get old value from main memory?

gfavor, 2023-10-15:
On Sat, Oct 14, 2023 at 8:09 AM fennecJ ***@***.***> wrote:

> Hart 0 Hart 1
> li t1, 3 li t3, 2
> li t2, 1
> (a) sw t1,0(s0) (d) lw a0,0(s1)
> (b) fence w,w (e) sw a0,0(s0)
> (c) sw t2,0(s1) (f) sw t3,0(s0)
>
> ...
>
> If Hart 0 and Hart 1 have their own write buffer,
> and the result of (c) haven't drain to main memory and appear in global
> memory order,
> may Hart1 not aware of write operation by Hart0's (c) and get old value
> from main memory?
>
The key is the initial statement "if the load (d) reads the valid 1".  If
that condition is true, then (a) must precede (f).  But if that condition
is false (e.g. because store (c) hasn't drained out and become globally
visible), then all bets are off.

Greg

fennecJ, 2023-10-15:
I see, that makes sense.
Thank you very much.

closed 2023-10-15 by fennecJ